microsoft/Azure-Kinect-Sensor-SDK #1471: Toothpick and Azure Kinect

vipelz, 2021-01-12:
I would like to purchase an Azure Kinect for a project.
    Before buying it, I would like to know what is the best distance at which Azure Kinect sees a toothpick with the depth module and with what quality?  Because a toothpick is pretty thin.  
   Could someone post some pictures of how the Kinect Azure sees the toothpick?
    Thanks

This is what I want to achieve:
    1) recognize the toothpick
    2) know where it is in space with respect to the fixed Azure Kinect positioned at the top facing down 
    3) grasp the toothpick with a robotic arm

qm13, 2021-01-21:
It is unlikely you will get reliable results on detecting objects as thin as toothpicks with Azure Kinect DK. That said you could work with our partners in the depth commercialization program to develop a Kinect based depth module tailored to detect toothpick sized objects.

vipelz, 2021-01-22:
Could you tell me more about depth commercialization program?
Thank you

L4Z3RC47, 2021-01-22:
> Could you tell me more about depth commercialization program?
> Thank you

+1

AshokPeddakotla-MSFT, 2021-02-01:
Same question posted on Stack Overflow: https://stackoverflow.com/questions/65755240/toothpick-and-azure-kinect-depth

tostercx, 2021-02-01:
Don't have a toothpick, but a roughly-toothpick-sized screwdriver gets picked up up to ~1m (using unbinned NFOV mode). You can see it's distorted in the depth camera tho - way thicker than it should be. The shape seems to be somewhat accurate up to ~40cm give or take. I don't think it's really built for this tho ^^;

[image]

closed 2021-02-09 by qm13