microsoft/Azure-Kinect-Sensor-SDK #1473: Issues with the extension of AKDK cable up to 5 meters

SimoSR, 2021-01-14:
We would like to extend the lenght of the camera cable up to 5 meters.

We have investigated two different ways: 

1. Single camera cable by adopting a USB 3.1 Gen 1 type C cable with active repeater (IC) o bring both power and data
2. Double camera cable by adopting a USB 3.1 Gen 1 type C to Type A cable for data + an USB 2.0 type A cable for additional power.

For the solution 1 we have tested this cable (https://www.cablematters.com/pc-1136-124-full-feature-active-usb-c-cable-with-4k-video-10-gbps-data-transfer-and-60w-power-delivery.aspx) that works fine if connected directly to a type C port on the PC motherboard. The problem is that we don't have a type C port available, we have used an expansion PCI-e card but we have experemented issues on corruped RGB stream due to the known compatibility issue with AS-Media controller (as described here  #1354 )

For the solution 2 we have tried to extend the two native cables provided with the camera:

- The 5 meters extension on the power cable (USB 2.0 type A) works fine, no issues are detected in providing the additional power to the camere.

- The issues come with the extension or the USB 3.1 Gen 1 data cable. In this case active cable is required to avoid loss of bandwith due to the lenght of the cable and the huge amount of data to be passed during the stream. We have tried this type of cable (https://www.amazon.it/CSL-ripetitore-Extention-amplificazione-amplificatore/dp/B00O1X82UC/ref=sr_1_6?adgrpid=86656707549&dchild=1&gclid=Cj0KCQiAqo3-BRDoARIsAE5vnaKXQLhra0BONUZvF_ntkr_OrHROmZDIWYrOj4oW9sYqih440dz8SpUaAk7hEALw_wcB&hvadid=407288793113&hvdev=c&hvlocphy=20561&hvnetw=g&hvqmt=b&hvrand=12166797773764512912&hvtargid=kwd-317206011106&hydadcr=15883_1812430&keywords=prolunga%2Busb%2B3.0%2B10%2Bmetri&qid=1607364954&sr=8-6&tag=slhyin-21&th=1).
No additional USB controllers are used in this case, the two USB cable are plugged directly to the USB 3.1 and 2.0 available on the motherboard.

In this case the behaviour of the camera when connected to the same PC is very unpredictable:

1. sometimes the device is correctly recognized and the stream of both RGB and depth are good even at high resolutions and framerates (no loss of data or lag detected).
2. other times the system is not able to detect the camera driver and the device in not available in the AKDK viewer due to the wrong configuration (uncorrect enumeration under device manager of the AKDK camera)
3. the order in which the two cables are plugged in seems to affect the capability of the system to corretly recognized the camera (for example the data cable has to be plugged after the power cable). This issue happes only if the data cable has been extend with the active cable.
4. when the PC is turned OFF and power ON the camera is no longer recognized and you need to re-plug the cables again. This issue happes only if the data cable has been extend with the active cable.

Can anyone provide hints to find a solutions to the issues reported above?

themancalledjakob, 2021-01-14:
We had similar issues with multiple cameras on various extensions. This sounds horrible, but we simply automated a relay to cut connection of both data and power simultaneously just before starting our application and had no trouble since then. I'm not really recommending to do this, but it does work stable for months now with not a single error.

qm13, 2021-01-14:
This cable is known to work https://www.amazon.de/gp/product/B073J92722/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1

SimoSR, 2021-01-26:
UPDATE
I'm experimenting issue with the cable connections.

**Configuration adopted**: 
- ADKD connected to the host PC via 2 cables (USB 3.0 for Data and USB 2.0 for power);
- Both cables are extended with enxtension cable of 5 meters;
- the data cable (USB 3.0) is an active cable to support the extension lenght;

If we plug the cables in the following order
1. plug the DC plug (USB 2.0);
2. plug the type C data cable.

the AKDK is correctly configured (fixed white light and device correctly enumerated in the viewer).

Otherwise if we plug before the Type C data cable (light goes ambre flashing do to low power) and then the DC plug, the light turn off and the device cannot be recognized y the host PC

Only if I remove and plug again the data cable the ligth turn white fixed and the device is correctly recognized by the host PC.

The same problem occurs when the host PC is powered off and the restarted: the system is not able to detect the AKDK connected and the only way to restore the connection is to remove the type C data cable and plug it again

Is there an explaination of why this happens only with an extension of the DC plug USB 2.0 cable?
With the native cable provided with the sensor this problem does not occur.

Is there any solution to avoid to manually replug the type C data cable every time the the PC is turned off and on?

SimoSR, 2021-01-29:
In addition to the issue reported above regarding the extension of the power cable, we are experimenting an issue also with 5 meter extension of USB A data cable.

We are using the following active cable extension  https://www.amazon.de/gp/product/B073J92722/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1 that you say it's known to work.

This the situation of device manager and Kinect viewer:

[image]

As you can see the device is correctly detected under Universal Serial Bus menu and under Cameras.
But when i open the Kinect Viewer and I try to refresh the devices i get the following error:

[ error    ] : RuntimeClassInitialize(). Failed to find color camera in open camera: 0xc00d36d5
[ error    ] : SUCCEEDED( Microsoft::WRL::MakeAndInitialize<CMFCameraReader>(&color->m_spCameraReader, (GUID *)container_id)) returned failure in color_create()
[ error    ] : color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()

Any idea how to solve these annoying problems?

qm13, 2021-01-29:
@SimoSR this may be related to a firmware issue that we are in the process of fixing. What happens when you use the supplied cables in the same configuration? Can you replicate the power issue? Same for the RGB camera.

The RGB camera issue usually occurs when you have another application access the RGB camera.

SimoSR, 2021-01-30:
@qm13 thank you for the reply, below a detailed description of the problem and how to replicate it

**ISSUE WITH POWER EXTENSION**

**Configuration adopted**

1. AKDK with both supplied cables plugged in
2. Extend the power cable with a 5 meters USB cable extension (no matter if it's acrtive or passive, 2.0 or 3.0)

**Describe the bug**
If the data cable is plugged in before the power cable the host PC is not able to detect the device

**To Reproduce**

1. connect the AKDK with the data cable (usb 3.0) to the host pc
2. the led is flashing amber due to the low power provided by the data cable
3. connect the AKDK with the power cable + the extension (5 meters)
4. the led is turned off (no light) and the host pc does not detect any device attached
5. unplug and replug the data cable
6. the led goes solid white
7. the host pc now correctly detects and enumerates the attached device
8. the RGB and depth stream can be started in the AKDK Viewer

The same happens when the pc is shut down and the AKDK remains connected; at the power on, the device remains in an unknown state (no light) and the data cable needs to be re-plugged.

**Expected behavior**
The plug order should not affect the capability of the camera to be detected and the manual re-plugging of the cable should not be required.
The same is expected at the restart of the system.

**Notes**
- This behavior has not been detected when no power extenson is adopted (only supplied cables), or the power extension is shorter.
- maybe this is due to a drop of current provided by the cable over long USB cable, but in this case i would expect that the AKDK cannot be detected in every situation due to the low power, instead everything goes well if the data cable is plugged later than the power one.

Any idea on this strange behaviour?

SimoSR, 2021-01-30:
**ISSUE WITH DATA EXTENSION**

**Configuration adopted**

1. AKDK with both supplied cables plugged in
2. Extend the data cable with a 5 meters USB 3.0 active cable extension 

Example of cable tested:

- https://www.amazon.it/gp/product/B08B8P1VFR/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
- https://www.amazon.it/CSL-ripetitore-Extention-amplificazione-amplificatore/dp/B00O1X82UC/ref=pd_all_pref_1?pd_rd_w=cTSIg&pf_rd_p=6ecfb5f1-c1b8-4d10-be3e-933e610ae3a5&pf_rd_r=FS16X0KAZPFTHCQTSP9T&pd_rd_r=bc7182b0-7da6-4436-8477-f5ee1d4d724f&pd_rd_wg=kqIUS&pd_rd_i=B00O1X82UC&psc=1

**Describe the bug**
The AKDK camera is correctly detected by device manager (Azure Kinect 4k camera and Depth camera present under USB devices) but the AKDK viewer is not able to connect to the device.

The following error log is given by the AKDK viewer:
[ error ] : RuntimeClassInitialize(). Failed to find color camera in open camera: 0xc00d36d5
[ error ] : SUCCEEDED( Microsoft::WRL::MakeAndInitialize(&color->m_spCameraReader, (GUID *)container_id)) returned failure in color_create()
[ error ] : color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()

Possible causes:

1. the color camera is used by another application or the app han no permission to access the color camera;
2. the container ID of the color camera has been changed by the USB cable extension and so AKDK sdk is not able to recognize a depth and color camera associated to the same device (due to the different container ID, as well described here by @wes-b https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/427) 

I tend to esclude the cause because no camera app is currently installed on the pc and the device is correctly detected by the AKDK viewer when connected through the supplied cables without any extension.

How to identify the effective cause of the issue and how to solve it?

SimoSR, 2021-02-05:
@qm13 please, can you re-open this issue? at the moment the problems are not solved.

greatwallet, 2021-03-08:
I know it's irrelevant with your issue, but how did you extend your power cable?

talhayaparr, 2021-07-08:
I have a same problem. Have you found any solution? @SimoSR

crypto-g245, 2021-07-08:
I am also facing the same issue @SimoSR 
Have you found any solution to the problem? 

#### Application:

- Using two kinects to reconstruct the scene 

#### Configuration:

- Extend the data cable up to 3 meters using an Active USB cable (no extenders). Used cable to connect the cameras - https://www.amazon.de/KIWI-Design-Oculus-Daten%C3%BCbertragung-Hellgrau-Hellgrau-3M/dp/B08BPDVBHF/ref=sr_1_22?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=Oculus%2Bconnection%2Bcable%2C%2B3%2Bm%2C%2Bhigh-

- Using original power cables supplied with the device
- Azure kinect SDK V 1.4.1 (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md)

#### Issues:
-  One of the cameras is streaming the data correctly. but the second camera would display the errors in the log as displayed in the below snapshot from Kinect Viewer and the camera stops working with the following errors in series. The device is recognized when connected. Once the camera is started in Kinect viewer, the device stops working after few seconds. 

-  Both the device and cables are working and there are no issues with them. 
    
[image]

- The camera stops working after this error and Azure Kinect viewer displays following errors. 

[image]
[image]
[image]
[image]
[image]

  
- Both the cameras are working fine with the original type-c to type-A cables. This issue doesn't occur with original cables.

closed 2021-10-06 by qm13