microsoft/Azure-Kinect-Sensor-SDK #1474: Force reinitializing cameras

jjoachim, 2021-01-14:
I apologize for the category error, but I wasn't sure what else to tag this with.

I have a complicated program where sometimes things go wrong and stuff doesn't close cleanly. My issue is that the cameras sometimes get stuck in a state where they simply refuse to be opened on a program restart, and the only thing that seems to work is either restart the computer or its even sometimes necessary to unplug the cameras from power and reconnect them. On program start, I open devices serially and get these errors (note program tries to open the camera 3 times with 2s intervals when exceptions are thrown before giving up, though all the "t=2442" are identical so not sure):

`
[2021-01-13 14:04:46.247] [critical] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2021-01-13 14:04:46.247] [error] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2021-01-13 14:04:46.247] [error] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[2021-01-13 14:04:48.260] [critical] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2021-01-13 14:04:48.260] [error] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2021-01-13 14:04:48.260] [error] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[2021-01-13 14:04:50.267] [critical] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2021-01-13 14:04:50.267] [error] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2021-01-13 14:04:50.267] [error] [t=2442] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
`

Is there a way to forcefully free whatever context/state the k4a is in and start fresh?

qm13, 2021-01-19:
You can reset all connected Azure Kinect devices from the command line

`AzureKinectFirmwareTool -Reset`

You can also reset a single Azure Kinect by specifying the serial number.

jjoachim, 2021-01-22:
Can this be done programmatically in C++? Is there a source file for this tool in the repo somewhere that I can use as an example?

L4Z3RC47, 2021-01-22:
> Can this be done programmatically in C++? Is there a source file for this tool in the repo somewhere that I can use as an example?

Can you not use system() to call the argument? https://stackoverflow.com/questions/8832326/how-can-i-execute-a-command-line-command-from-a-c-program

jjoachim, 2021-01-22:
Yeah we can use system(), but it's hard to get return/exception feedback on commands like that. I'd prefer a more direct c-api if it exists, but I can make the system() way work in the mean time I think.

jjoachim, 2021-01-26:
Tested this solution on a faulty system, worked perfect. Will use system() for now to invoke the fix, but would like an api or some kind of instruction on how to do it myself. Closing, thanks for the help.

closed 2021-01-26 by jjoachim