microsoft/Azure-Kinect-Sensor-SDK #1478: Add an option for compressed depth streams (png supports 16-bit grayscale)

kdquine, 2021-01-16:
Hi! My lab has been working with the new camera, but one frustration we've had is the size of the depth streams, which are, well, uncompressed.  After a bit of investigation trying to find a good lossless compression that works in-frame for videos, we found an option that appears to work in compressing the depth frames as png images and tagging the depth streams in the mkv container as MPNG, which video players to play the 16-bit depth streams directly! (We set header->biCompression = 0x474E504D).

catid, 2021-01-25:
Curious how this compares to my work: https://github.com/catid/Zdepth

kdquine, 2021-01-29:
It turns out compressing pngs (with libpng at least) takes a good chunk of time (50ms?) and we were silently dropping frames.  Would need a faster solution than the one I tried.  ZDepth looks pretty fast!