microsoft/Azure-Kinect-Sensor-SDK #1479: Need a body tracking sdk that depends on libk4a1.4-dev

seigeweapon, 2021-01-18:
**Describe the bug**

Current libk4abt1.0 depends on libk4a1.3-dev, with which we have experienced unstable capturing (looks like #1194). This happens almost 100% when using a 5 meter usb extension cable. When using libk4a1.4-dev, the frame dropping still happens, but it is a warning instead of fatal error.

Now, in our project, I have to use libk4a and libk4abt both. To avoid libk4a1.3-dev's instability, I have to use libk4a1.4-dev. But libk4abt1.0 requires using libk4a1.3-dev.

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

qm13, 2021-01-19:
There is a release in the works that addresses this issue.

seigeweapon, 2021-02-25:
alright, waiting for the good news

closed 2021-02-25 by seigeweapon