microsoft/Azure-Kinect-Sensor-SDK #1480: Question about k4a_image_create_from_buffer

logic03, 2021-01-18:
k4a_result_t k4a_image_create_from_buffer	
(k4a_image_format_t 	format,
int 	width_pixels,
int 	height_pixels,
int 	stride_bytes,
uint8_t * 	buffer,
size_t 	buffer_size,
k4a_memory_destroy_cb_t * 	buffer_release_cb,
void * 	buffer_release_cb_context,
k4a_image_t * 	image_handle 
)	
——————————————————————————————————————————
关于这个函数的参数，如果缓冲区是uint8_t类型的，说明给这幅图像的数据只能是0-255，意思就是不能创建unsigned short(0-65535)类型的深度图像数据，那么第一个参数就不能是K4A_IMAGE_FORMAT_DEPTH16.请问这个API是不是实现的有问题呢?
我的目的是先把深度图以png的形式存储到本地,然后在另一个程序里面读取这张16位的深度图,通过k4a_image_create_from_buffer这个函数创建一个k4a_image_t类型的数据。
————————————————————————————————————————————
The first parameter to this function cannot be K4A_IMAGE_FORMAT_DEPTH16 if the buffer is of type UINT8_T, which means that the data given to the image can only be 0-255. This means that no unsigned short(0-65535) can be created.Is there a problem with the implementation of this API?

My goal is to store the depth map locally as a PNG, then read the 16-bit depth map in another program and create a k4a_image_t with the function k4a_image_create_from_buffer.

RoseFlunder, 2021-01-18:
Hi,

if you can specify depth16 as image format.
The buffer is always uint8 (byte) and the image format specifies how to interpret that buffer.
Depth16 means that two bytes will be concatenated and interpret as a uint16 value.
Hope that is kind of understandable to you.

logic03, 2021-01-19:
> Hi,
> 
> you can specify depth16 as image format.
> 
> The buffer is always uint8 (byte) and the image format specifies how to interpret that buffer.
> Depth16 means that two bytes will be concatenated and interpret as a uint16 value.
> Hope that is kind of understandable to you.

那么当我想要调用k4a_image_create_from_buffer这个函数把png图像里面的像素值赋值给一个k4a_image_t类型的对象的时候,buffer_release_cb和buffer_release_cb_context参数应该怎样设置呢？直接设置成：
k4a_memory_destroy_cb_t* buffer_release_cb = NULL;
void* buffer_release_cb_context = NULL;
这样可以吗？
——————————————————————————————————
So when I want to call the k4a _ image _ create _ from _ buffer function to assign the pixel value in the png image to a k4a _ image _ t type object, how to set the buffer _ release _ cb and buffer _ release _ cb _ context parameters ? Set directly to :
k4a_memory_destroy_cb_t* buffer_release_cb = NULL;
void* buffer_release_cb_context = NULL;
It is OK?

closed 2021-02-22 by qm13