microsoft/Azure-Kinect-Sensor-SDK #1481: Body Tracking not working on AMD CPU's / Motherboard's?

Shureshure, 2021-01-18:
Hi
Unfortunately i switched from Intel to an AMD Ryzen 9 5900X
Now the Problem:

The Azure Kinect Viewer works flawlessly, but the Body Tracking Viewer it does not work.
I get into the program, but for some reason the 3D Visualization Window opens only after five or ten Minutes.
The Camera transmits the Image, but the skeleton is completely missing (only a red point can be seen).

There are problems with AMD CPU's / Motherboard's? are there any solutions?

Many thanks for the answer

RoseFlunder, 2021-01-18:
I don't think so.
I use a Ryzen 3700X (Zen 2) on a X570 Chipset with an NVidia 1080 and everything works

There is a known issue regarding the new Nvidia 3000 Series if you also have a new GPU by chance?
At the moment we can't use any Nvidia 3000 Series (Ampere GPU) with the current body tracking sdk version

qm13, 2021-01-19:
@Shureshure does your new AMD computer have an NVIDIA GPU? Is so what NVIDIA GPU does it use?

Shureshure, 2021-01-19:
Thank you for the fast answer
Yes i have a rtx 3090. I was so fixated on amd that I did not get the idea that it could be due to the graphics card.
Now I see that I'm not the only one who has problems with it.
The problem doesn't seem to have been solved. Is there any news about a patch?

qm13, 2021-01-20:
Ok. We are working on an update that provides support for the NVIDIA 3xxx range of GPU cards. We are planning to get it out (barring issues) by mid February.

asergaz, 2021-01-30:
Related thread in Microsoft Q&A: https://docs.microsoft.com/en-us/answers/questions/238927/index.html

UnaNancyOwen, 2021-02-22:
@qm13 How is the update release going? I'm looking forward to it! 😄

zirkeltraining, 2021-02-25:
Same here, we're anxiously hoping for the fix, otherwise we need to return that expensive new Machine….

Shureshure, 2021-02-25:
@qm13 mid February 2022? or 2023?  How long it will take? it would be nice to know how the situation is.

telhabilal-arbisoft, 2021-03-01:
Do we have any updates regarding the 30 series update release?

guobenyu1976, 2021-03-02:
@qm13 i also want to know the date update. if it could be release this week, we could wait for it. otherwise we need to borrow/buy another 20 series computer to let the project go on.

sgrafbus, 2021-03-09:
@qm13 and responsible MS folks: Can we please get an update on the status or your latest plans to support the NVIDIA 3xxx range of GPU cards for the Azure Kinect body tracking? We have a core development project relying on the use of the feature in combination with the 3xxx GPU performance, so would appreciate a transparent update on what we can expect by when, please. Thank you!

zirkeltraining, 2021-03-15:
we're still waiting and would be very happy about an update on this issue.

YongsunLee, 2021-03-17:
Is this problem solved?
If not, I would like to know the official answer when it will be resolved.

qm13, 2021-03-19:
Added ONNX DirectML execution environment support (default environment). Now possible to run on AMD and Intel GPUs on Windows. Performance is dependent on the GPU.

closed 2021-03-19 by qm13

vertex798, 2021-04-20:
> Added ONNX DirectML execution environment support (default environment). Now possible to run on AMD and Intel GPUs on Windows. Performance is dependent on the GPU.

AMD Rizen 5300, RTX 3090.
What steps should I take to use this feature?

ZodiacWind, 2021-12-13:
> > Added ONNX DirectML execution environment support (default environment). Now possible to run on AMD and Intel GPUs on Windows. Performance is dependent on the GPU.
> 
> AMD Rizen 5900, RTX 3090. What steps should I take to use this feature?

Same question. I cannot make it with Razen 5950X and RTX 3080Ti. Could anybody provide some guidelines?

MarioTed, 2023-01-30:
> Same question. I cannot make it with Razen 5950X and RTX 3080Ti. Could anybody provide some guidelines?

Same question. Idk to how to do it.

tinywolf3, 2023-06-11:
3060, 3070, 3080, same here. please fix it.