microsoft/Azure-Kinect-Sensor-SDK #1484: Azure Kinect slowed down & Body Tracking not working on CPU 8 core Ryzen 7 3700x & GPU rtx 3070

hayeonp95, 2021-01-22:
Hi Everyone,

I have been using Azure Kinect and PC with CPU of intellcore i7 and GPU of rtx 2060 super and it had been working perfectly fine.

So I recently changed to pc with high specification which is CPU of 8 core Ryzen 7 3700x and GPU of rtx3070.
But after that I have been having problems which it takes as long as 15mins for it to start sample program and it can’t even identify skeleton.

I was wondering where the problem is coming from cpu or gpu or the main board?

Would there be any possibility that it is due to not having an update that provides support for the gpu 3070?

Thank you!

RoseFlunder, 2021-01-22:
Hi,

its because of the new nvidia 3000 series.
See this issue for further information:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1481#issuecomment-763288112

qm13, 2021-02-22:
Duplicate of https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1481

closed 2021-02-22 by qm13