microsoft/Azure-Kinect-Sensor-SDK #1485: Question about GreenScreen

logic03, 2021-01-22:
请问为什么要使用绿色的布（绿屏）进行标定的实验呢？我认为是绿屏可以吸收深度相机的红外发射器发出的光，被绿屏吸收了这个光就无法反射回相机，从而使得绿屏部分的深度为0。我这样理解对吗？
————————————————————————————————————————————
Why do we use green cloth ( green screen ) for calibration experiments ? I think the green screen can absorb the light emitted by the infrared emitter of the depth camera, which is absorbed by the green screen and cannot reflect back to the camera, so that the depth of the green screen part is 0. I understand that, right ?

tostercx, 2021-01-22:
There's no need for an actual green screen for calibration. It's a demo application that simulates a green screen effect with the depth camera - everything close to the camera will be drawn in view but the background will remain static.

logic03, 2021-01-22:
> There's no need for an actual green screen for calibration. It's a demo application that simulates a green screen effect with the depth camera - everything close to the camera will be drawn in view but the background will remain static.

意思就是绿色的布是不能够吸收深度相机发射出的红外光，是这样吗？
_________________________________________________________________________
The green cloth does not absorb the infrared light emitted by the depth camera, right?

tostercx, 2021-01-22:
You don't need green cloth for calibration! It works with no cloth.

closed 2021-02-09 by qm13