microsoft/Azure-Kinect-Sensor-SDK #1488: I use pyAzureKinect to Get Calibration but it always null, could you tell me how to get calibration's Intrinsic

UEBoy2019, 2021-01-23:
'
pyK4 = pyKinectAzure(modulePath)
pyK4.device_open(0)
config = pyK4.config
pyK4.device_start_cameras(config)
config.color_resolution = _k4a.K4A_COLOR_RESOLUTION_1080P
#config.depth_mode = _k4a.K4A_DEPTH_MODE_WFOV_2X2BINNED
config.depth_mode = _k4a.K4A_DEPTH_MODE_NFOV_UNBINNED
cali = None
pyK4.getDepthSensorCalibration(calibration=cali)
#Traceback (most recent call last):
# File "D:/UE4Protect/pyKinect/pyKinectAzure/examples/Test3DReConstruct.py", line 25, in <module>
# pyK4.getDepthSensorCalibration(calibration=cali)
#File "../pyKinectAzure\pyKinectAzure.py", line 573, in getDepthSensorCalibration
#self.device_get_calibration(self.config.depth_mode, self.config.color_resolution, calibration)
#File "../pyKinectAzure\pyKinectAzure.py", line 280, in device_get_calibration
#_k4a.VERIFY(self.k4a.k4a_device_get_calibration(self.device_handle,depth_mode,color_resolution,calibration),"Get calibration failed!")
# OSError: exception: access violation writing 0x0000000000000080

o3d_Intrinsic = o3d.camera.PinholeCameraIntrinsic
o3d_Intrinsic.width = cali.depth_camera_calibration.intrinsics.parameters.codx
o3d_Intrinsic.height = cali.depth_camera_calibration.intrinsics.parameters.cody
o3d_Intrinsic.fx = cali.depth_camera_calibration.intrinsics.parameters.fx
o3d_Intrinsic.fy = cali.depth_camera_calibration.intrinsics.parameters.fy
o3d_Intrinsic.cx = cali.depth_camera_calibration.intrinsics.parameters.cx
o3d_Intrinsic.cy = cali.depth_camera_calibration.intrinsics.parameters.cy
print(o3d_Intrinsic, cali.depth_camera_calibration.intrinsics.parameters)
`

zapaishchykova, 2021-01-28:
following, also want to know how to retrieve the Kinect intrinsics!

zapaishchykova, 2021-02-09:
here is to find the calibration intrinsics: build from source azure sdk using this https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md and then build a calibration_info.exe https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration which outputs the intrinsics. Hope this helps!

closed 2021-02-22 by qm13

wangmiaowei, 2021-06-04:
Well my output is as follows:
{"CalibrationInformation":{"Cameras":[{"Intrinsics":{"ModelParameterCount":14,"ModelParameters":[0.5014757513999939,0.49704870581626892,0.49203813076019287,0.49193161725997925,4.72473669052124,3.2092227935791016,0.17059645056724548,5.0558595657348633,4.775780200958252,0.88649415969848633,0,0,0.0001238211989402771,5.87981094213319E-6],"ModelType":"CALIBRATION_LensDistortionModelBrownConrady"},"Location":"CALIBRATION_CameraLocationD0","Purpose":"CALIBRATION_CameraPurposeDepth","MetricRadius":1.7399997711181641,"Rt":{"Rotation":[1,0,0,0,1,0,0,0,1],"Translation":[0,0,0]},"SensorHeight":1024,"SensorWidth":1024,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}},{"Intrinsics":{"ModelParameterCount":14,"ModelParameters":[0.506722092628479,0.51457756757736206,0.47892501950263977,0.63847380876541138,0.40386343002319336,-2.6028842926025391,1.5118778944015503,0.28427526354789734,-2.4207212924957275,1.4354244470596313,0,0,-4.1651688661659136E-5,0.00089407846098765731],"ModelType":"CALIBRATION_LensDistortionModelBrownConrady"},"Location":"CALIBRATION_CameraLocationPV0","Purpose":"CALIBRATION_CameraPurposePhotoVideo","MetricRadius":0,"Rt":{"Rotation":[0.99999916553497314,-0.00080411869566887617,-0.0010013245046138763,0.00089394015958532691,0.99564725160598755,0.09319741278886795,0.00092202425003051758,-0.093198232352733612,0.995647132396698],"Translation":[-0.032077357172966,-0.0019188573351129889,0.0041174949146807194]},"SensorHeight":3072,"SensorWidth":4096,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}}],"InertialSensors":[{"BiasTemperatureModel":[-0.010616951622068882,0,0,0,0.021280378103256226,0,0,0,0.010436063632369041,0,0,0],"BiasUncertainty":[9.9999997473787516E-5,9.9999997473787516E-5,9.9999997473787516E-5],"Id":"CALIBRATION_InertialSensorId_LSM6DSM","MixingMatrixTemperatureModel":[0.9957544207572937,0,0,0,-0.001605906174518168,0,0,0,0.0072632487863302231,0,0,0,-0.001582253840751946,0,0,0,1.0105862617492676,0,0,0,0.00074907898670062423,0,0,0,0.007269588764756918,0,0,0,0.00076094019459560513,0,0,0,0.99488413333892822,0,0,0],"ModelTypeMask":16,"Noise":[0.00095000001601874828,0.00095000001601874828,0.00095000001601874828,0,0,0],"Rt":{"Rotation":[-0.00031475935247726738,0.11594212800264359,-0.99325591325759888,-0.99999648332595825,0.0025804587639868259,0.00061811070190742612,0.0026347208768129349,0.99325263500213623,0.11594090610742569],"Translation":[0,0,0]},"SecondOrderScaling":[0,0,0,0,0,0,0,0,0],"SensorType":"CALIBRATION_InertialSensorType_Gyro","TemperatureBounds":[5,60],"TemperatureC":0},{"BiasTemperatureModel":[-0.01494935154914856,0,0,0,-0.045461028814315796,0,0,0,-0.0642952173948288,0,0,0],"BiasUncertainty":[0.0099999997764825821,0.0099999997764825821,0.0099999997764825821],"Id":"CALIBRATION_InertialSensorId_LSM6DSM","MixingMatrixTemperatureModel":[0.99677795171737671,0,0,0,-8.6717009253334254E-5,0,0,0,-0.0031037656590342522,0,0,0,-8.7303553300444037E-5,0,0,0,0.99007153511047363,0,0,0,0.00033213524147868156,0,0,0,-0.003125463379547,0,0,0,0.00033221009653061628,0,0,0,0.98985821008682251,0,0,0],"ModelTypeMask":56,"Noise":[0.010700000450015068,0.010700000450015068,0.010700000450015068,0,0,0],"Rt":{"Rotation":[0.00087367423111572862,0.11046438664197922,-0.99387967586517334,-0.99999916553497314,0.0010460560442879796,-0.00076279003405943513,0.00095539272297173738,0.99387955665588379,0.11046520620584488],"Translation":[-0.051448330283164978,0.0037394696846604347,0.0014938798267394304]},"SecondOrderScaling":[0,0,0,0,0,0,0,0,0],"SensorType":"CALIBRATION_InertialSensorType_Accelerometer","TemperatureBounds":[5,60],"TemperatureC":0}],"Metadata":{"SerialId":"000565500712","FactoryCalDate":"2/17/2020 11:32:29 AM GMT","Version":{"Major":1,"Minor":2},"DeviceName":"AzureKinect-PV","Notes":"PV0_max_radius_invalid"}}} 

How can I get the intrinsic matrix? It is a little confused for me to understand these information