microsoft/Azure-Kinect-Sensor-SDK #1489: one PC, 4 kinect  sensor,  The maximum number of body tracking supported?

seoyeo, 2021-01-23:
I have 4 kinect sensor, use body tracking in Unity, now i can tracking 6 body . But I want to be able to tracking 15-20 body, what should i do? thanks!

L4Z3RC47, 2021-01-23:
You can run each Kinect sensor on separate networked computers and use multiplayer HLAPI to handle merging all of the tracked skeletons. Note that you would then need to ensure that each cameras covered area does not overlap otherwise you may get duplicated skeletons. Additionally each camera zone will still only be able to track up to 6 skeletons. This is probably the simplest way to increase tracked bodies

seoyeo, 2021-01-25:
thanks for your reply！
Are there other ways to increase body tracking count？

closed 2021-02-09 by qm13