microsoft/Azure-Kinect-Sensor-SDK #1496: What is the status of this SDK?

dbruning, 2021-01-28:
I have a product based on Kinect for Xbox One and I'm keen to upgrade it to support the Kinect Azure camera, but I'm worried about the lack of development in this repo.

There don't seem to have been any code commits in the past 6 months, and no PRs closed since May 2020. 
Questions are still being answered, but it feels a bit like the project has been all but abandoned for any development.
Can anyone give an update or talk about guaranteed support for this product? I don't want to be burned again with another discontinued product.

Further, can anyone comment on what this announcement about a [collaboration to develop ToF cameras with Leopard Imaging](https://www.leopardimaging.com/1-27-2021-leopard-imaging-and-microsoft-collaborate-to-develop-tof-cameras/)?

bumbastic, 2021-01-31:
I have the same issue. I have a product ready, that would require 20-50 Kinects in the first year, this year. But can I trust that Microsoft will continue to deliver?

brontesprocessing, 2021-02-04:
That's pretty interesting for me too

dbruning, 2021-02-07:
[Azure depth platform](https://devblogs.microsoft.com/azure-depth-platform/) is now a thing - there are a couple of recent blog posts on there, including 

My interpretation is: "we have some truly great ToF technology but outside of HoloLens we don't really know what to do with it. KinectDK seemed like a good idea at the time, but it's too much work to support (too many different use cases) and not our core business. So, we're going to license the IP to some hardware companies instead". 

You probably can't comment on that, but I really would like to know:
1. Who is likely to produce a device that is closest to the spirit of the Kinect v2 (inexpensive, widely available)
2. What are their roadmaps/timelines to get that product to market?
3. Is the KinectDK going to live on as a reference implementation, or is it deprecated now?
4. Is the KinectDK ever going to be available worldwide, rather than just in some cherry-picked countries? 

It's so frustrating that Kinect v2 was such a groundbreaking and awesome product, and here we are nearly a decade after its release with nothing comparable in price/performance.

qm13, 2021-02-08:
Thank you for your feedback. Azure Kinect DK is part of Microsoft’s Azure depth platform program, aimed at democratizing cloud connected 3D vision, we are working to establish an ecosystem of camera makers, ISVs, and SIs, to build solutions based on our 3D ToF and Azure platforms. For more information please visit : https://aka.ms/azure3d.

If you are interested in learning more, partnering or becoming a customer please email azure3d@microsoft.com.

dbruning, 2021-02-09:
Thanks for the response. I will email to connect directly, but could you please answer the central question of this issue: is this SDK still supported? Put another way: can we release products based on this SDK (and the Azure Kinect) camera, knowing that it will continue to be supported into the future? Or do we need to wait for other partners in this ecosystem to release commercial hardware?

NickAtMixxus, 2021-02-09:
So what does it mean?
In the long run, sounds good if it leads to more cameras.  
In the short run, for now, is there a Kinect for Azure device? Will it ever be sold to the public?
If not, can't help think it's really sad. Unlike Titanic, Kinect V2 didn't sink, it was sunk. Yet still, my Kinect V2 is here working splendidly and in theory I could develop many apps for it but if it gets broken I can't buy a new one.
One wonders if the responsible bosses at Microsoft are aware that Microsoft created one of the last decades most amazing technical device, a camera that fast and accurate transforms movements to data. The implemantations for Kinect are endless, especially in health apps and it should really be a Microsoft thing.

qm13, 2021-02-09:
All - yes, both SDKs are still supported and you can expect to see updates to both in the next few months.

The other more important from a commercialization POV is we are working on an update to Sensor SDK aligned with commercialization to support the commercial modules and cameras. The current Sensor SDK makes some assumptions around AKDK which need to be broken. Expect to make some code changes to use the updated Sensor SDK for commercial devices. This version of the SDK will also support AKDK.

dbruning, 2021-02-09:
Thanks for that update, it's great news. Can I just check I understood correctly:

- AKDK = Azure Kinect Developer Kit (the Azure Kinect camera hardware)
- Sensor SDK = Azure Kinect Sensor SDK (this repo)
- The Sensor SDK will be broadened in the next few months to work with upcoming commercial hardware, not just the AKDK
- The AKDK hardware itself will also be updated in the next few months

qm13, 2021-02-09:
Almost correct. The last point is the AKDK hardware will get a firmware update in the next few months. This will be included in an update to the Sensor SDK (note that this is not the update that will add commercial sensor support, that is later in the year).

SimoSR, 2021-02-10:
Hi Quentin, thank you for the information provided, they are all welcome!

A couple of additional questions that i think will be usefull for all of us :)

1. Can you explain better what do you mean with "commercial sensor support" that is coming later this year?
2. Do you refer also to the AKDK camera (HW) or just to the SDK (SW)?
3. Will it be an extension of the available markets to buy (at the moment limited to USA, China, Japan, UK and Germany)?
4. For how many years do you expect to maintain the AKDK on the market?
5. When i look at the list of the current disclosed partners (sick, Analog Device, Leopard Imaging, ...) that are going to develop their own camera/module, it seems that they aim to replicate just the ToF camera module (without the mapping with the RGB sensor): do you know if it will be an effective commercial replacement to the AKDK camera (RGB-D sensor with both color and ToF image)?

DrMaxP, 2021-02-12:
Can you say any more about this... "The AKDK hardware itself will also be updated in the next few months". Having just bought my fourth sensor wondering if something interesting odds on its way.

DrMaxP, 2021-02-12:
Are you talking about more than just firmware updates? Also the blog post on improved depth performance in more challenging environmental conditions was interesting. Are these the kind of improvements we might expect to come to the AKDK?

qm13, 2021-02-22:
@DrMaxP these are coming in future depth modules / cameras developed by vendors in partnership with Microsoft depth team. Some of this (particularly depth engine improvements) will made available for the current Azure Kinect DK through an upgrade to the Sensor SDK.

dimitrijevic, 2021-03-07:
Could you comment on the price point of those future depth modules / cameras please...AKDK with the same price it had when it came out a year ago and having new devices come out in the near future with better specs for lesser or even equal price is bad...anything coming in terms of AKDK price cuts or 'sweeting the pot' with extra software or azure services included would make it better, or having something like a development edition HL2, being able to exchange devices for the latest and best spec one calling it something like Kinect 365 would be best. Will the new devices be available in the Microsoft Store and when? TIA

bumbastic, 2021-03-08:
Kinect is awesome! For almost 10 years now, the 360 Kinect Party App, has been the center of events, and always a big hit, at my kids birthday parties. My 11 year old boy just found my old kinect 360 and kinect 2 PC adapters, and Odyssey HMD. He put it together by him self, and now he is showing off in VR chat, using the kinect for full body tracking. Myself, I'm using the k4a for fast 3D mesh reconstruction, in Scan and Plan projects with industrial robots performing surface processing.

closed 2021-04-13 by qm13