microsoft/Azure-Kinect-Sensor-SDK #1498: AKDK connected to the Intel USB 3.1 eXtensible Host Controller are not enumlated.

tokawa-ms, 2021-01-29:
**Describe the bug**
Devices connected to the Intel USB 3.1 eXtensible Host Controller are not enumlated.

**To Reproduce**
1. Plugin the Azure Kinect to USB 3.0 Port and start the laptop
2. Check in device manager the USB Port.
3. The device appears to be recognized correctly.
[image]
4. Run [Repro Code](https://github.com/tokawa-ms/k4a_serial_read)
5. No device detected.
[image]

**Expected behavior**
The device is properly enumlated.
By running Repro Code, the device will be recognized and the serial will be displayed.

**Logs**
[k4a.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/5894449/k4a.log)

**Laptop (please complete the following information):**
 - OS with Version: Windows 10 Version 20H2 (build 19042.746)
 - SDK Version:  Azure Kinect SDK v1.4.1
 - Firmware version: 
 Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
  
  (I captured AKDK firmware versions by the other PC)

- CPU : Intel Core i7 8700
- Chipset : Intel H370
- Graphics : NVIDIA GeForce RTX 2060

tokawa-ms, 2021-01-29:
**Additional Info**
I can use AKDK as a USB Web Camera by Photo App. (I can capture the video stream from AKDK with Windows Photo App.)
But I can't connect AKDK with Azure Kinect Sensor SDK / Body Tracking SDK.

qm13, 2021-02-01:
@tokawa-ms do you have a task in the background e.g. k4aviewer that is accessing the Azure Kinect depth camera? The depth camera cannot be accessed by more than one application at the same time.

tokawa-ms, 2021-02-02:
@qm13 I have no process or task in the background. I'm accessing from only one process.

How can I capture any verbose log for this kind of issue?

qm13, 2021-02-22:
https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#collecting-logs

closed 2021-04-13 by qm13