microsoft/Azure-Kinect-Sensor-SDK #1507: can't compile C:2039 error cinttypes.h

drhalftone, 2021-02-15:
Never had a problem compiling my code before using Qt with VS2017; however, I now have brand new Windows laptop with VS 2019 Community and Qt 5.12.2 and now pops up an error I have never seen before:

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(23): error C2039: 'imaxdiv_t': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(23): error C2873: 'imaxdiv_t': symbol cannot be used in a using-declaration
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(25): error C2039: 'imaxabs': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(25): error C2873: 'imaxabs': symbol cannot be used in a using-declaration
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(26): error C2039: 'imaxdiv': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(26): error C2873: 'imaxdiv': symbol cannot be used in a using-declaration
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(27): error C2039: 'strtoimax': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(27): error C2873: 'strtoimax': symbol cannot be used in a using-declaration
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(28): error C2039: 'strtoumax': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(28): error C2873: 'strtoumax': symbol cannot be used in a using-declaration
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(29): error C2039: 'wcstoimax': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(29): error C2873: 'wcstoimax': symbol cannot be used in a using-declaration
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(30): error C2039: 'wcstoumax': is not a member of '`global namespace''
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\cinttypes(30): error C2873: 'wcstoumax': symbol cannot be used in a using-declaration

drhalftone, 2021-02-16:
I was able to fix the issue by commenting out all references to the cinttypes.h header, which brings up the obvious question why are the references there in the first place?

drhalftone, 2021-02-16:
What is the solution?

qm13, 2021-02-16:
@drhalftone How is this related to Azure Kinect?

drhalftone, 2021-02-16:
They are your header files (k4a/*.h) that are causing the problem.  I deleted the "#include <cinttypes>” calls and then everything compiles fine.

qm13, 2021-02-17:
@drhalftone From your comments you imply that this project - using QT and Azure Kinect - used to compile ok on VS2017 but after upgrading to VS2019 no longer compiles? What happens when you compile a Azure Kinect sample (that does not use QT)?

closed 2021-04-13 by qm13

DISAPPEARED13, 2023-12-29:
it's weird that I have to commenting the reference to solve the problem, but it does work. Might be the complier version is the problem. But why?