microsoft/Azure-Kinect-Sensor-SDK #1508: k4abt_simple_3d_viewer.exe can not be run in RTX3070 laptop graphc card

guobenyu1976, 2021-02-16:
I just bought a new asus air2 (RTX 3070 laptop graphic card) to replace the old thinkpad. The sdk, cuda, cudnn is new. 

1, Azure Kinect SDK(1.4.1) and viewer is ok. device is ok.
2, k4abt_simple_3d_viewer.exe can be run in cmd line as "k4abt_simple_3d_viewer.exe CPU"
3, device and k4abt_simple_3d_viewer.exe is ok in my old thinkpad

but, if i double click k4abt_simple_3d_viewer.exe in asus air2 notebook, the command window will appear, but picture window can not appear.

because it can be run in CPU mode, but can not be run in GPU mode.

guobenyu1976, 2021-02-16:
the system information is as below:
[image]

[image]

RoseFlunder, 2021-02-16:
Known issue and according to Microsoft a patch comes soon:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1481#issuecomment-763288112

qm13, 2021-02-22:
Duplicate of https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1481

closed 2021-02-22 by qm13

guobenyu1976, 2021-03-03:
Hi, qm13, when will the exe be updated?

when i met the problem, you said it is going to update in mid February.
So i wait the update and did other part of the project.
Now it is March. and the exe did not update yet.
the project has to solve it. We have to wait for the update or change the hardware.

May i know the planned date to update the exe?
if it is going to update this week or next week, i am going to wait for the update.
if it met some unexpected problem, it is hard to expect update in March. I am going to discuss with my boss, and change the hardware.

thanks very much!!