microsoft/Azure-Kinect-Sensor-SDK #1510: body track DLL doesn't manage its dependencies; missing manifest, DLL load path

diablodale, 2021-02-21:
The body tracking dll `k4abt.dll` is missing DLL path handling for its own dependencies. This results in a failure to load k4abt.dll and cascades up to parent DLLs/EXEs.

It appears k4abt.dll lacks code to manage its own DLL load path and/or a manifest. Without these, Windows follows a well defined series of steps to find body tracking dependent DLLs. And in common cases of DLLs, it fails.

Many customers do not create single monolithic applications. Therefore, we create DLLs...just like K4A created two DLLs. The K4A DLLs need to play well in these scenarios.

### Setup

* Windows 10 current updates
* K4a SDK v1.4.1
* K4bt SDK v1.0.1
* Visual Studio 2019 Community v16.8.4

### Repro and Description

1. Create a Windows program EXE
2. Enhance that EXE so it can have new features through dynamically loading DLLs. (This is the common "plugin" scenario). Ensure this DLL feature is loaded dynamically...not static...not at program start. It is loaded only through LoadLibrary().
3. Create a DLL to be the Kinect Body Tracking feature.
4. Install your EXE in a directory like "c:\program files\myprogram"
5. Install your plugin in a different directory like "c:\myplugins"
6. Copy all body tracking dependent DLLs into c:\myplugins. These DLLs are: k4a.dll, k4bt.dll, onnxruntime.dll, cudnn64_7.dll, cublas64_100.dll, etc.
7. Run your program and dynamically load your feature

### Result

The DLL feature fails to load. The cause is it can not find any of the body tracking depedent DLLs.
Using Sysinternals process monitor, I can see the loader trying to find these DLLs and it can not. It searchs the windows os folders, path, etc.... can't be found.

The cause is within k4abt.dll itself. It has no manifest, therefore the windows loader doesn't look in the current or any other specific directory. In addition k4abt.dll doesn't call `SetDllDirectoryW(the_k4abt.dll_dir)`. If it did, it would look in the same directory as itself.

There is also an interesting and annoying failure. The feature DLL can successfully load k4a.dll and k4abt.dll by using its own manifest and/or hooking it via a delayload dll. Yet, even though the feature DLL loaded k4a.dll and it is in memory...that doesn't equate to the k4a.dll that is needed by k4abt.dll. When I use Sysinternals process monitor, I see the feature DLL successfully load k4.dll, then I see k4abt.dll load, then I see k4a.dll fail to load because the body tracking dll is separately loading it and fails due ot above issues.

### Expected

Feature loads. No errors.

### Workarounds

None known so far.

1. The feature DLL can not list these dependent DLLs in its manifest. Why? Because the feature manifest only handles loads for its direct dependencies (e.g. k4a.dll). It is not used for its dependencies dependencies (e.g. cudnn64_7.dll).
2. The feature DLL can not call `SetDllDirectoryW(the_k4abt.dll_dir)`. That API only changes the search path for its own dependencies. It is not used for its dependencies dependencies.

vladalx, 2021-03-08:
Configuring the dll search path(s) is the responsibility of the programmer not of the library.
The VerifyDependencies functions only checks for the installation of two critical components, CUDA and CUDNN.

diablodale, 2021-03-08:
Hello. You seem to be a new user or perhaps a bot.
I believe you are errant in your post. I must point out that you did not read my information above as I documented correct usage of DLL search paths and that `VerifyDependencies()` is incorrectly looking in CWD rather than the DLL search path.

Unless you can provide specific information, I believe your post is not relevant to this conversation.

scotthsu98052, 2021-08-12:
@qm13  this is a feature request. Current BTSDK doesn't support application isolation (side-by-side assemblies). The standard search order for Windows applications is used for loading linked dlls.

diablodale, 2021-08-12:
@qm13, that is misleading. While LoadLibrary might be using standard path, the function `VerifyDependencies` IS NOT.
This is distinct.
Two actions.

1. VerifyDependencies is a separate and independent function in the k4abt that checks for the presence of certain files.
2. Later, various DLLs are loaded by k4abt using Windows standard paths

It is (1) of which I'm reporting this bug. Not (2). This is a recurring misunderstanding by your team.

scotthsu98052, 2021-08-13:
Hi @diablodale,
_Re: Perhaps part of the issue is in your function VerifyDependencies(). Are you blindly using currentworkingdirectory? Or the directory of the master EXE?_ 

No, VerifyDependencies is not doing what you thought. It is a just wrapper call of Windows API [SearchPath](https://docs.microsoft.com/en-us/windows/win32/api/processenv/nf-processenv-searchpatha). 
`return SearchPath(nullptr, fileName.data(), ".dll", 0, nullptr, nullptr) != 0;`

The lpPath parameter is NULL, so the default search order is similar to how system loads DLLs. 
If you explicitly use LoadLibrary, then this check won't work well for you, but this is what current design is.

diablodale, 2021-08-18:
Ah ha! You have admitted the problem.

`SearchPath` api *does not* follow the same rules and have the same functionality of the DLL load APIs. 👎

> The SearchPath function is not recommended as a method of locating a .dll file if the intended use of the output is in a call to the LoadLibrary function. This can result in locating the wrong .dll file because the search order of the SearchPath function differs from the search order used by the LoadLibrary function. If you need to locate and load a .dll file, use the LoadLibrary function.

It follows different paths, ignores apis like `SetDllDirectory()` and `AddDllDirectory()`, and as officially written by your own corporation...should not be used to find DLLs, resources, etc.

This is related to https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#using-body-tracking-sdk-with-unreal and its general topic of "finding required resources (dll,onnx,etc.) for k4a and k4abt SDKs.

qm13, 2022-03-21:
Done

closed 2022-03-21 by qm13

diablodale, 2022-05-25:
@qm13, Done in what version? For customers to verify unseeable code changes, we need to know the exact version in which is was potentially fixed so that customers can verify, escalate non-fix, etc.