microsoft/Azure-Kinect-Sensor-SDK #1519: Body Tracker mid-February 2021 update delayed?

Brekel, 2021-03-03:
There have been mentions of a Body Tracker SDK update being right around the corner, scheduled for mid February 2021, bringing NVidia 3000 support (and hopefully other improvements).

Is there any word on the delay and/or a new estimate?
Numerous people are eagerly awaiting it so even a beta would help tremendously.

qm13, 2021-03-04:
Sorry for the delay. Adding RTX 30xx support threw up some unexpected issues. We plan to build the final release candidate Friday and publish the release early next week.

Brekel, 2021-03-04:
That's great to hear, thanks for sharing that info!

closed 2021-03-04 by Brekel

jaymimc, 2021-03-11:
> Sorry for the delay. Adding RTX 30xx support threw up some unexpected issues. We plan to build the final release candidate Friday and publish the release early next week.

Has the RTX 3xxxx support update been published now then?

Jeremboo, 2021-03-12:
And will it be accompanied by support for other graphics cards? I'm waiting for some information about the AMD GPUs support :)

Blargstomp404, 2021-03-12:
> 
> 
> Sorry for the delay. Adding RTX 30xx support threw up some unexpected issues. We plan to build the final release candidate Friday and publish the release early next week.

New with Azure Kinect, is there a link to try out the rc for rtx 3xxx cards?

Brekel, 2021-03-19:
Looks like the new MSI installer just hit the website :)
https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-download

AswinkarthikeyenAK, 2021-03-26:
Hi Guys, 
Has anyone tested the NVidia 30xx series for body tracking in Ubuntu 18.04 by any chance? It will be really helpful to know this information.

Thanks