microsoft/Azure-Kinect-Sensor-SDK #1520: Camera calibration

zhaocheng-hub, 2021-03-03:
Hello, my depth image and color image don't match very well. How do I need to calibrate my camera.

diablodale, 2021-03-03:
https://docs.microsoft.com/en-us/azure/kinect-dk/use-calibration-functions

zhaocheng-hub, 2021-03-04:
@diablodale  , Hello, 
I'm referring to the depth image and color image obtained by capture, and use them to calibrate the camera calibration data. The  'transformation_color_image_to_depth'  image I got is not very good.   

I want to find a calibration method similar to "https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration"

diablodale, 2021-03-04:
That link you provided is a good resource. The procedure and technology to create calibration matrices is well documented throughout the internet. I recommend you google "how to create camera calibration matrices".

Also, consult the OpenCV API reference to learn how to use their APIs to sample many frames/images, tag them, and then create calibration matrices from them. The more samples, and the more exact you are tagging those samples, may lead to a set of matrices that better meets your needs. You could use another tech/api, but OpenCV is free and well understood.

zhaocheng-hub, 2021-03-04:
@diablodale   Hello, thank you for your advice.

I have tried to use 'opencv', 'ROS' and' matlab 'for calibration, but the calibration structure I got is not good, probably because of my lack of ability. I hope there is an official way to use and learn

FilipePires98, 2021-03-29:
> @diablodale Hello, thank you for your advice.
> 
> I have tried to use 'opencv', 'ROS' and' matlab 'for calibration, but the calibration structure I got is not good, probably because of my lack of ability. I hope there is an official way to use and learn

Any progress on this?

zhaocheng-hub, 2021-04-10:
@FilipePires98   Hello,
I haven't made any progress so far. Have you made any progress?

FilipePires98, 2021-04-21:
> @FilipePires98 Hello,
> I haven't made any progress so far. Have you made any progress?

Hello,
It seems to me that Microsoft is comfortable with their factory calibration quality for the Azure Kinects and has no intention to provide tools to improve the intrinsic calibration.
For my research, I have decided to accept this as a system constraint, however in the future it should definitely be explored in greater detail with tools such as the one you mentioned above, adapted for K4A.

zhaocheng-hub, 2021-06-07:
@FilipePires98  Hello，
The project now provides a calibration method [https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration_registration](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration_registration)

FilipePires98, 2021-06-15:
> @FilipePires98 Hello，
> The project now provides a calibration method https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration_registration

Hello @zhaocheng-hub , thank you for your comment. I was not aware of this recent update, this is great news.

closed 2021-08-16 by qm13