microsoft/Azure-Kinect-Sensor-SDK #1521: Open source BT SDK

rfilkov, 2021-03-05:
Please consider again open-sourcing the Body Tracking SDK, unless there is something really secretive and proprietary in it. This may help workaround the major issues, because many users would try to improve it in many directions. It may ease and help the team at the same time. The blockers with the NVidia and other GPUs would have been solved months ago, if the source was open.

qm13, 2021-03-09:
There are no plans to open source the Body Tracking SDK. The DNN model is inferenced using ONNX Runtime which is open-sourced.

rfilkov, 2021-03-09:
OK. Thank you for the sincere response! It was expected.
Unfortunately, the body tracking ruined this otherwise great sensor.

closed 2021-03-09 by rfilkov