microsoft/Azure-Kinect-Sensor-SDK #1523: Azure Kinect Camera Update to highest specification

hayeonp95, 2021-03-10:
Hi, 

It has been 10months since the azure kinect camera has updated, so I was wondering when would you do update to the highest specification? 

I'm currently using azure kinect and developing fitness training at home and what concerns me is its discontinuation as no updates has been done for 10 months. 

Thank you!

dimitrijevic, 2021-03-10:
Think you should read [this](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1496) 

I’d like to know if those future devices are going to be able to sync with the AKDK also, or should I purchase as many of those as I can now, given that my project’s goal was to use multiple Kinect devices in the end...

I’m guessing the new devices will have a lesser price point given LG announced mass production of their camera modules based on MS IP two days ago...

Brekel, 2021-03-10:
@dimitrijevic do you have a link to LG's announcement? Sounds interesting!

> I’m guessing the new devices will have a lesser price point given LG announced mass production of their camera modules based on MS IP two days ago...

closed 2021-04-13 by qm13