microsoft/Azure-Kinect-Sensor-SDK #1526: Encounter the error when build sdk in ubuntu16.04

lulindeng, 2021-03-12:
Hi,

I try to build sdk from the source, I followed this [reference steps](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1327#issuecomment-678100229) to install and build.

When I run 'ninja', I encountered the error:
`[39/151] Building CXX object examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o
FAILED: /usr/bin/c++   -I/usr/include/opencv -Isrc/sdk/include -I../include -O2 -g -DNDEBUG -fPIE -fvisibility=hidden -fvisibility-inlines-hidden   -Wall -Wextra -Wno-missing-field-initializers -Werror -std=gnu++11 -MD -MT examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o -MF examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o.d -o examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o -c ../examples/green_screen/main.cpp
../examples/green_screen/main.cpp: In function ‘Transformation stereo_calibration(const k4a::calibration&, const k4a::calibration&, const std::vector<std::vector<cv::Point_<float> > >&, const std::vector<std::vector<cv::Point_<float> > >&, const Size&, const Size&, float)’:
../examples/green_screen/main.cpp:505:116: error: could not convert ‘16648’ from ‘int’ to ‘cv::TermCriteria’
                                        cv::CALIB_FIX_INTRINSIC | cv::CALIB_RATIONAL_MODEL | cv::CALIB_CB_FAST_CHECK);
                                                                                                                    ^
[39/151] Building CXX object tests/DepthTests/UnitTest/CMakeFiles/depth_ut.dir/depth_ut.cpp.o
ninja: build stopped: subcommand failed.
`

Seems like something wrong with my opencv, I installed Opencv using `sudo apt install libopencv-dev`.

Does anyone encountered this error before? How can I fix this error?

Deepest thanks to your reply!

HlibKazakov2000, 2021-03-17:
Hello,
Azure Kinect doesn't support Ubuntu 16.04. Consider updating your software to Ubuntu 18.04.

This error could be caused because your libopencv-dev version is outdated. Ubuntu 16.04 packages have version 2.x of the opencv library which is considered an old one. You can try to install a newer version from github or from the official site. But I would recommend you to update your system

closed 2021-03-17 by HlibKazakov2000

lulindeng, 2021-03-17:
> Hello,
> Azure Kinect doesn't support Ubuntu 16.04. Consider updating your software to Ubuntu 18.04.
>
> This error could be caused because your libopencv-dev version is outdated. Ubuntu 16.04 packages [...]

Hello, 

Thanks for your reply! I understand it would be better to update the system, but I have other projects which need ubuntu 16.04. The problem has been solved by build latest OpenCV version from the source. and now the `ninja` works fine.

I also have another question, After this building process, Can I use the body tracking SDK now? or I need to follow [Kinect official documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-download) to install the body tracking SDK? Because when I try `sudo apt install k4a-tools` or `sudo apt install libk4abt1.3-dev`, it cannot work and I have no idea to solve this problem.

Deepest thanks for your reply!

HlibKazakov2000, 2021-03-17:
k4a-tools and libk4abt1.3-dev packages are not present in the Microsoft Ubuntu 16.04 repository. That's why it doesn't work.
```Can I use the body tracking SDK now?```
It is hard to say because you are using a deprecated version of Ubuntu

lulindeng, 2021-03-17:
> k4a-tools and libk4abt1.3-dev packages are not present in the Microsoft Ubuntu 16.04 repository. That's why it doesn't work.
> `Can I use the body tracking SDK now?`
> It is hard to say because you are using a deprecated version of Ubuntu

Thanks for you prompt reply! 

I'll try it when I receive the Azure Kinect :) Thanks again!