microsoft/Azure-Kinect-Sensor-SDK #1527: Azure Kinect Sync Not Working

cfortney92, 2021-03-12:
**Describe the bug**
I'm attempting to set one Azure Kinect as a master and the other as the sub. The setup is as follows:
(Master) connected via USB to PC (i7-6820HK + GTX 1080), 7m 3.5mm TRS from SYNC OUT 
(Sub) connected via USB to PC (i9-10900k + RTX 2080 Super), other end of TRS cable into SYNC IN 

**Expected behavior**
To see both devices listed in the Kinect Viewer, but each PC is only showing their own device connected to it. Each PC is able to start their own device individually, but they are not synced even when one is started as sub on its own PC (first) and master started on its own PC (last). If I'm understanding the documentation correctly, I should be able to see both devices listed in Kinect Viewer on the same PC and not need to start them individually on their respective PC's. 

Both computers match all versions between them:
 - OS with Version: Windows 10 10.0.19042 Build 19042
 - SDK Version:  1.4.1
 - Firmware version:    
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

RoseFlunder, 2021-03-13:
No that is exactly how it should be.
You need to start them individually on the respective PCs.
Subordinate kinects first because they wait with capturing until master is started.

cfortney92, 2021-03-13:
> No that is exactly how it should be.
> You need to start them individually on the respective PCs.
> Subordinate kinects first because they wait with capturing until master is started.

Thanks! The documentation hadn't mentioned that so I was misunderstanding. The views of the 2 cameras aren't automatically combined in the viewer after they're both synced, is there an easy method for going about this? I'm interested in using both Kinects at the same time for increased accuracy in body tracking but I'm a novice when it comes to coding.

closed 2021-03-18 by qm13