microsoft/Azure-Kinect-Sensor-SDK #1531: Availibility of Kinect DK sensor

fdechaumont, 2021-03-17:
Hi,

Thanks for the kinect DK: it's awesome !

I am sorry my question is not related to the SDK but related to the hardware.

The sensor is currently out of stock, do you know when it will become available again ?
Is it still restricted to a few countries ?

Thanks

Fab

qm13, 2021-03-18:
Azure Kinect DK is currently in short supply due to increased demand and the world-wide consumer electronic components shortage. We expect this to remain the case until early July.

fdechaumont, 2021-03-19:
Thanks a lot qm13 !

I am always scared in the discontinuity in production. This is reassuring.

FYI, I am currently using the kinect 2 to create this open source project : [https://livemousetracker.org/](https://livemousetracker.org/) 
It is dedicated to the live 24/7 observation of the animals. It's basically a real time tracker.
This was about impossible to do without your sensors. Warm thanks for that.

I hope to switch the Live Mouse Tracker community to the new sensor. I am very much interested by the synchronization between cameras. Once again, great job !

On the website, you will see that the community is also in countries that can't have access to the Azure Kinect DK Hardware (Specifically France). Do you know if the country limitation will still exists in July ?

I think new coming system will need 2 cameras each, and I expect the community to order for about 80 cameras in 2021.

Best

PierrePlantard, 2021-06-08:
Dear @qm13,
Do you have a more precise expected date for when the cameras will be available again?
Best

fig8studio, 2021-06-30:
I was just informed by a Microsoft rep that the Kinect DK is now on 6-8 months backorder, meaning Jan-Mar 2022... ooph!

qm13, 2021-07-08:
@fig8studio this is not correct. We are seeing delays of 1-2 months depending on the size of the order.

We have moved to a managed sales process with all 5 stores (US, China, Japan, UK, and EU) in "Call to order" with call details. The store sales person will capture your immediate needs and 3 month forecast. This information will be used to assign devices and forecast demand for each SKU. Once devices have been assigned for you immediate needs you will get a call back from the store sales team to place and pay for your order. Your order should be delivered within a week.

fig8studio, 2021-07-08:
@qm13 I really do appreciate that you're shedding light on the situation to the GH community - definitely more so than the MS/Azure site itself.
I'm not saying you're wrong (not at all), but I'm confident in what I was told by both a Microsoft rep as well as their 'managed salespeople' (I was ping-ponged between two companies over the course of 4 days). Seriously, I just need 3 of them, and I feel like I'm being vetted for the purchase of ex-military equipment...!
My comments are not for the sake of complaining, but for everyone reading this thread. I was able to purchase 1 Kinect DK before they ran out, and I was really impressed by it. But since then, it's been a nightmare to implement into my research due to their unavailability. Instead of waiting for an ever-changing restock date (I was told March/June/July, and now '6-8 months after you're approved'), I genuinely wish I opted for another device. I could have used the past 4 months on code development rather than chasing up a non-existent product.
Again, no qualms with you qm13, hopefully others get to read this and see that getting a Kinect DK may (or may not) be so easy.

mellinger, 2021-07-08:
> @fig8studio this is not correct. We are seeing delays of 1-2 months depending on the size of the order.
>
> We have moved to a managed sales process with all 5 stores (US, China, Japan, UK, and EU) in [...]

I just called the 1-800 number on the product page and read them this verbatim, they had no idea what it was in reference to and said they'd been getting calls about it all day.

Abzdefgxyj, 2021-07-12:
I am trying to order a quantity in the hundreds and have told the 1-800 sales reps this on multiple occasions. Would love a different number to call! 1-2 months is heartening.

submorphic, 2021-07-24:
Similar experience with 1-800 number as others. Salesperson did not know any details and simply took my contact info and will notify me when available.  We're looking for several dozen devices, but also now going back to re-evaluate Intel 455 which seem to be readily available.  The AK was superior in our initial evaluations, but we can not wait 6-8 months for more to become available and will likely have to move on.

qm13, 2021-07-24:
All, it is taking longer to roll out the program than initially anticipated. The UK and US stores should be up to speed by early next week with the China, Japan and German stores following a week later (due to content localization requirement). If you are still having problems starting August 1st please reply to this issue and I will look into it.

Waffleicious, 2021-08-02:
> All, it is taking longer to roll out the program than initially anticipated. The UK and US stores should be up to speed by early next week with the China, Japan and German stores following a week [...]

Hello I am trying to purchase just one Kinect Azure and the Microsoft rep just told me they have no idea when they will be back in stock... Any updated information would be amazing

Abzdefgxyj, 2021-08-02:
Checked this morning and they still had no idea unfortunately

brontesprocessing, 2021-08-03:
Are there any updates regarding availability?

Abzdefgxyj, 2021-08-04:
I was told on that sales line that they would be out of stock until October 2022.  @qm13 do you you know if the managed sales process still uses the basic sales phone number found on the store page? 1-866-425-4709 (for US)?

From your description it sounded like you guys had been receiving orders throughout this time but I don't see how that's possible via the number above, as they've been consistently telling me they're just out of stock (which is the same info as on the store page).

qm13, 2021-08-04:
@Abzdefgxyj This is not correct. I have passed your comment on to the lead for the US Store. The sales person is supposed to capture how many Kinects you need now and forecast for next 3 months. This info is used in a managed sales process to assign production against needs. Once production is assigned to your need someone from the assisted sales team will reach out to you to place a firm order. It is currently taking 1-2 months to address needs,

Abzdefgxyj, 2021-08-04:
@qm13 I saw your previous comments, so I'm greatly encouraged that this is still the case. I'll keep trying this week and next and hopefully will see some success.

millsrandy, 2021-08-06:
I don't know if it's ok to post this here or not, (please let me know if I'm in the wrong here), but I have two DK units (one barely used (took it out of the box and hooked it up a couple times) and one brand new sealed in the box) that I'd like to get rid of.  Please DM me if interested, they are immediately available.

Abzdefgxyj, 2021-08-13:
@qm13 I'm being told that October is the availability, and that Microsoft can't fill the need in-house for anything larger than 5 units. I've been told to reach out to partnerdirectory to find someone to fill this need.

I don't think the sales people have gotten the memo or that there's a managed sales process yet. Please advise which number to call if that's not the case.

TheShoes, 2021-08-19:
I am just trying to buy 1.  Is my only hope to keep visiting the Microsoft page to hope to see them in stock one day?

submorphic, 2021-08-19:
I’ve spoken to sales twice now and they could not take an order but did collect info for a quote. They promised the quote back in 24 hours on on - never received anything back either time.  On the second call they acknowledged having the prior request and escalated. Still nothing.  Will call again today.

qm13, 2021-08-19:
@TheShoes this will not work. Azure Kinect DK has been moved to a managed sales process and all stores are set to "Call to Order"
@submorphic you should not have been told they will get back to you in 24 hours, please accept my apologies. The process takes time from collecting your interest to buy thru getting a quote. You will not get a quote until we have allocated production against your interest.

TheShoes, 2021-08-19:
@qm13 oh thats a little misleading on the website. 
should i call this number then  1-866-425-4709
is there even a chance of getting 3 units this month?
just wondering if i am going to have to look at a different solution

qm13, 2021-08-19:
Yes, please call that number. Small orders like yours are taking between 2-4 weeks to fill.

submorphic, 2021-08-20:
> I’ve spoken to sales twice now and they could not take an order but did collect info for a quote. They promised the quote back in 24 hours - never received anything back either time. On the second call they acknowledged having the prior request and escalated. Still nothing. Will call again today.

Just updating for others.  Chatted with agent who found my two prior call logs.  They went offline to talk to manager and "backend" team. Came back and said I would have a quote in 3 hrs.  Well, it's the next morning and nothing.  Will call in again this morning and try one more time.  We're looking for a small number asap, but I asked for a quote for 25, and we're planning on 500+ over next 18-24 months if all goes as planned.  The very first agent I spoke to weeks ago was in the know....everyone else has been clueless.

Chris45215, 2021-08-20:
I daresay that Microsoft should increase the price of the sensors in order to reduce demand slightly and encourage buyers to only purchase what they need immediately; and to increase the profit margins enough to justify a larger commitment of Microsoft's resources to the project. Change the price from the current $400 to a temporary $550 or $600 - that is much more reasonable than the current Ebay price of $1000.

submorphic, 2021-08-20:
Just called store again....told nothing available right now in any volume. Again, asked to put in a quote and watch for email - some may be available next month.  Sigh.

TheShoes, 2021-08-20:
@qm13 is there any other number i can try.  I called the store and they said everything in the next shipment has been spoken for and i have to call back September 30th to try and get an update.

JimCheong, 2021-08-31:
I am selling mine in ebay for those that can't wait for $800.  Free shipping to Continent of US only.  1st come 1st serve

https://www.ebay.com/itm/255114864826

fig8studio, 2021-08-31:
@JimCheong - that's a pretty shitty thing to do

JimCheong, 2021-08-31:
> @JimCheong - that's a pretty shitty thing to do

It is a pretty shitty thing for you to say.  This is purely supply demand Adam Smith modern macroeconomy at work.  A shitty thing to do is when people needs it, I am not willing to part with it

fig8studio, 2021-08-31:
> It is a pretty shitty thing for you to say. This is purely supply demand Adam Smith modern macroeconomy at work. A shitty thing to do is when people needs it, I am not willing to part with it. It is not like I am hoarding. This is my personal unit for development

facepalm

JimCheong, 2021-08-31:
> > It is a pretty shitty thing for you to say. This is purely supply demand Adam Smith modern macroeconomy at work. A shitty thing to do is when people needs it, I am not willing to part with it. It is not like I am hoarding. This is my personal unit for development
> 
> facepalm

So, you rather that I just auction it off then?  Just so you understand.  Here is the result of an auction.  $900 and they have to wait for auction to end.  Time wasted.  "Those that can't wait" means if you can, go order it from Microsoft and as for you, if you got nothing positive to say, STFU, some golden rules in life.

https://www.ebay.com/itm/393457292064?hash=item5b9be1d320:g:bIcAAOSwgodg-Pnz

fig8studio, 2021-08-31:
> So, you rather that I just auction it off then? Just so you understand. Here is the result of an auction. $900 and they have to wait for auction to end. Time wasted. "Those that can't wait" means if [...]

MSRP $400, Academic Discount $360... here you are on a forum of programmers and researchers trying to exploit a supply shortage by selling one for double the price of a brand new one. The fact that you're just selling one means you'd be exploiting someone just trying to get started with the Kinect, as opposed to a lab/company needing dozens if not hundreds.

And just think about what you wrote: "if you got nothing positive to say, STFU..." maybe read your own words?

I'm done with you. And I hope you get a scam buyer for your scam listing.

JimCheong, 2021-08-31:
> > So, you rather that I just auction it off then? Just so you understand. Here is the result of an auction. $900 and they have to wait for auction to end. Time wasted. "Those that can't wait" means if you can, go order it from Microsoft and as for you, if you got nothing positive to say, STFU, some golden rules in life.
> 
> MSRP $400, Academic Discount $360... here you are on a forum of programmers and researchers trying to exploit a supply shortage by selling one for double the price of a brand new one. The fact that you're just selling one means you'd be exploiting someone just trying to get started with the Kinect, as opposed to a lab/company needing dozens if not hundreds.
> 
> And just think about what you wrote: "if you got nothing positive to say, STFU..." maybe read your own words?
> 
> I'm done with you. And I hope you get a scam buyer for your scam listing.
> 
> *Edit: In response to shady forum members like JimCheong, if you're beginning/testing the Azure Kinect DK and just need it for a limited time, I have one that you can use for free. The only limits are: 1. You live in the US, 2. You pay for shipping (free pickup around PA/NJ/CT corridor), and 3. You agree to keep the camera in clean condition. If interested, PM me.

Look, I bet you never own stocks with the stock market before, or sold a home you love before.  You seems to be pissing at folks that sell stocks or their home in market condition where the price doubled.  What is wrong with you?

JimCheong, 2021-08-31:
> I am selling mine in ebay for those that can't wait for $800. Free shipping to Continent of US only. 1st come 1st serve. Will ship priority mail.
> 
> https://www.ebay.com/itm/255114864826

Item sold - thanks to free market supply meets demand, another developer can continue their work

MrMizz, 2021-08-31:
hey @JimCheong, I bet you don't write code. get off these forums.

JimCheong, 2021-08-31:
> hey @JimCheong, I bet you don't write code. get off these forums.

And I bet you fill up gas at the gas station every now and then and scream that someone scam you off because the price of gas changed.

fig8studio, 2021-08-31:
@JimCheong - in the interest of keeping this page dedicated to updates re/ the Kinect DK, let's call a truce and stop the back and forth.

mellinger, 2021-09-03:
So, uhm, getting back to Kinect availability... 

@qm13 I just got off the phone for maybe the 10th time and they took my information for a quote only after I heavily insisted. I was told the quote would do me no good until October 2022, they had none of your optimism around a 2-4 week order fulfillment (for an order of 7). 

I get the sense your ability to get a Kinect right now is heavily influenced by who you are. I don't want to be competing with the Scatters and the Triggers of the world, this is extremely frustrating. I'd love if a member of your team could reach out, maxellinger@gmail.com

JimCheong, 2021-09-03:
I called them too to order one for myself after I sold mine couple of days ago.  I choose the option for them to notify me around oct 2022 as production catches up with demand.  I will then have the opportunity to order and they say then it will be a week for it to arrive.  The current market price is about $990 to $1600 and I believe there is are a few hoarders selling at that price for multiple unit in eBay and AliExpress.

omidb, 2021-09-14:
@qm13 hi, can you please confirm that the Azure Kinect DK is backorders by end of Oct 2022? I just called and that what they told me.

Ankitjaiswal1201, 2021-09-30:
> @qm13 I really do appreciate that you're shedding light on the situation to the GH community - definitely more so than the MS/Azure site itself. I'm not saying you're wrong (not at all), but I'm confident in what I was told by both a Microsoft rep as well as their 'managed salespeople' (I was ping-ponged between two companies over the course of 4 days). Seriously, I just need 3 of them, and I feel like I'm being vetted for the purchase of ex-military equipment...! My comments are not for the sake of complaining, but for everyone reading this thread. I was able to purchase 1 Kinect DK before they ran out, and I was really impressed by it. But since then, it's been a nightmare to implement into my research due to their unavailability. Instead of waiting for an ever-changing restock date (I was told March/June/July, and now '6-8 months after you're approved'), I genuinely wish I opted for another device. I could have used the past 4 months on code development rather than chasing up a non-existent product. Again, no qualms with you qm13, hopefully others get to read this and see that getting a Kinect DK may (or may not) be so easy.
> 
> *Also, I should clarify that I'm operating within the US market. Meaning, my situation may not be relevant for all markets.
> 
> **FWIW, I have also been testing my work using Intel RealSense devices. The Kinect DK produces FAR cleaner results, and its body tracking applet is superior to that of Intel. However, and this is a big however, Intel's products are available for immediate shipping. Between the D455 and/or the L515, I've been able to get comparable results for my research purposes (with some data clean up). I've been holding out for the DK as this would save a lot of coding hassle.

Looks like I have to do the same. Waiting for 1 and half months. Initially told it will be available in 1 month and when I spoke yesterday, they said waiting time is 6 months. This is for Germany.

fig8studio, 2021-09-30:
It's a bit late now, but FWIW, even Intel is EOL'ing their RealSense line (at least most products now, maybe more soon).
Must be something much larger than just Microsoft. Starting to consider StereoLabs' ZED line more seriously...

Ankitjaiswal1201, 2021-09-30:
> It's a bit late now, but FWIW, even Intel is EOL'ing their RealSense line (at least most products now, maybe more soon).
> Must be something much larger than just Microsoft. Starting to consider StereoLabs' ZED line more seriously...

I will also start looking for ZED... Have to see how good the depth details are.

omidb, 2021-10-08:
@qm13 Can you please share contact to some of these 3rd parties?

AswinkarthikeyenAK, 2021-12-01:
Hi @qm13,
Any updates when the camera might be back in stock?

Thanks

filipp-backvall, 2022-02-13:
Curious if anyone has had any shipments lately? I got excited with the tech and AK hardware but reading this thread does worry me. Being a beginner in the CV space I just need a single item but don't want to end up with abandoned product.

JohnHardy, 2022-02-14:
I have similar issues here.  I have an *ideal* use-case for the Azure Kinect DK in an industrial product we manufacture, but nobody at MS seems to be able to offer them for sale.

Does anyone here have a contact that actually knows about the product and where it can be purchased?

qm13, 2022-02-14:
Supply and demand and approaching equilibrium and we expect to reactivate online purchases of up to 5 units by mid March. In the meantime you can call the Microsoft store (listed on the purchase page for your region) and request a quote for devices. Depending on the region and size of the order you will receive your devices in 2-6 weeks.

JohnHardy, 2022-02-14:
Thanks for your reply!  Do you know who we can speak to in the UK? 

It would be great to be able to have a contact who can help with supply chain planning, moqs, end of life, etc.

At the moment I can't find anyone who knows anything about the product line - which is a real shame because it's completely brilliant!  Buying a dev kit here and there only gets us so far.

filipp-backvall, 2022-02-14:
Thanks for the update @qm13, I've already requested a quote here in Sweden and hope to be up and running within a few weeks. 🙌

petrasvestartas, 2022-02-16:
If someone is selling Azure Kinect DK please contact me by mail petras.vestartas@epfl.ch

RMichaelPickering, 2022-03-21:
@qm13 Can you please provide an update on availability, in this case for Canada -- presumably I'd use the US store to order, is the online availability expected soon? Also is there a third party supplier working on just the 7 microphone array part of the AK and if so could you please provide more information and a contact? Finally, is there any update on the 3d party cameras yet? TIA!

RinCat, 2022-03-21:
@RMichaelPickering In January, I ordered 3 x Kinect DKs from MS store Canada agent, still not shipped.

RMichaelPickering, 2022-03-21:
@RinCat Thanks for that information, I'll take some small encouragement that at least you were able to order the AK DK from MS Canada store -- I didn't even think it was available in Canada yet!!

qm13, 2022-03-21:
The US store is accepting orders in CAD for AKDKs and shipping to Canada. That said the US store is considerably backordered and will take time to fulfill all the orders.

petrasvestartas, 2022-03-21:
@qm13  is there any possibility to order one kinect to Switzerland?
I called a month ago to microsoft, they put the order but they never replied ever since...

qm13, 2022-03-21:
@petrasvestartas the Japan, UK and EU stores are moving back to unmanaged sales this week. That means the return of the "Buy" button on the AKDK page for those stores. You should be able to then place an online order for a single Kinect in Euros to be shipped to Switzerland.

petrasvestartas, 2022-03-22:
@qm13 

Thank you for a response. Could you please post a link for EU Microsoft? I am always redirected to us site.

qm13, 2022-03-22:
https://azure.microsoft.com/de-de/services/kinect-dk/

[5 more replies]