microsoft/Azure-Kinect-Sensor-SDK #1535: Proper way to read synchronous images

fdechaumont, 2021-03-19:
Hi,

I am trying to read 2 synchronized cameras in cpp. My goal is to feed a windows' NamedPipe to a java program.
My problem is that if a frame is not read, I drop a frame, and then any further image will be unsynchronized between the cameras.

I guess I am not using the SDK properly.

I only need IR and depth, but It seems that for sync purposes I do need to grab the color on the master ?
Camera are connected with a classic audio jack
I setup my cameras like this:

```cpp
	// SUBORDINATE
	k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	config.color_resolution = K4A_COLOR_RESOLUTION_OFF;
	config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config.wired_sync_mode = K4A_WIRED_SYNC_MODE_SUBORDINATE;
	config.subordinate_delay_off_master_usec = 160; // Recommended min of 160uS to avoid multiple cam interference

	// MASTER
	k4a_device_configuration_t config2 = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config2.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	//config.color_resolution = K4A_COLOR_RESOLUTION_OFF;
	config2.color_resolution = K4A_COLOR_RESOLUTION_720P; // Master has to grab visual image for sync purposes...
	config2.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config2.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config2.wired_sync_mode = K4A_WIRED_SYNC_MODE_MASTER;
```

I start the cameras as follow:

```cpp
	printf("Starting camera 1 (left)\n");
	if (K4A_RESULT_SUCCEEDED != k4a_device_start_cameras(device, &config))
	{
		printf("Failed to start device. Problem with sync cable ?\n");
		if (device != NULL)
		{
			k4a_device_close(device);
			
			exit(1);
		}
	}
	
	printf("Starting camera 2 (right)\n");
	if (K4A_RESULT_SUCCEEDED != k4a_device_start_cameras(device2, &config2))
	{
		printf("Failed to start device2\n");
		if (device2 != NULL)
		{
			k4a_device_close(device2);
			exit(1);
		}
	}
```

Then I have a thread grabbing the images.
If I have any frame drop, exectution will pass through the "drop frame" message at the end of this portion of code.
For the next grabs, my cameras are no longer synchronized.

```cpp
DWORD WINAPI ContinuousCaptureThread(LPVOID lpvParam)
{
	printf("Starting continuous capture thread...\n");

	const int32_t TIMEOUT_IN_MS = 1000;
	k4a_capture_t capture = NULL;
	k4a_capture_t capture2 = NULL;
	int frameNumber = 0;

	using namespace std::chrono;
	milliseconds msStart = duration_cast< milliseconds >( system_clock::now().time_since_epoch() );
	long msLast = 0;

	for (;;)
	{

		k4a_image_t depth_image2 = NULL;
		k4a_image_t ir_image2 = NULL;

		// Get a depth frame from device 2
		switch (k4a_device_get_capture(device2, &capture2, TIMEOUT_IN_MS))
		{
		case K4A_WAIT_RESULT_SUCCEEDED:
			break;
		case K4A_WAIT_RESULT_TIMEOUT:
			printf("Timed out waiting for a capture 2\n");
			break;
		case K4A_WAIT_RESULT_FAILED:
			printf("Failed to read a capture 2\n");
			return 0;
		}

		k4a_image_t depth_image = NULL;
		k4a_image_t ir_image=NULL;
		
		// Get a depth frame from device 1
		switch (k4a_device_get_capture(device, &capture, TIMEOUT_IN_MS))
		{
		case K4A_WAIT_RESULT_SUCCEEDED:
			break;
		case K4A_WAIT_RESULT_TIMEOUT:
			printf("Timed out waiting for a capture\n");
			break;
		case K4A_WAIT_RESULT_FAILED:
			printf("Failed to read a capture\n");
			return 0;
		}

		depth_image2 = k4a_capture_get_depth_image(capture2);
		ir_image2 = k4a_capture_get_ir_image(capture2);
		k4a_capture_release(capture2);
		
		depth_image = k4a_capture_get_depth_image(capture);
		ir_image = k4a_capture_get_ir_image(capture);
		k4a_capture_release(capture);
		
		
		if ((depth_image != NULL) && (ir_image != NULL) && (depth_image2 != NULL) && (ir_image2 != NULL))
		{
				
			// Write depth image data to reply
			uint8_t *depth_image_buf = k4a_image_get_buffer(depth_image);
			size_t depth_buf_size = k4a_image_get_size(depth_image);
				
			uint8_t *depth_image_buf2 = k4a_image_get_buffer(depth_image2);
			size_t depth_buf_size2 = k4a_image_get_size(depth_image2);

			uint8_t *ir_image_buf = k4a_image_get_buffer(ir_image);
			size_t ir_buf_size = k4a_image_get_size(ir_image);

			uint8_t *ir_image_buf2 = k4a_image_get_buffer(ir_image2);
			size_t ir_buf_size2 = k4a_image_get_size(ir_image2);

			EnterCriticalSection(&CriticalSectionForCaptureCopy);
				
			memcpy(&captureBuffer[0], depth_image_buf, depth_buf_size);
			memcpy(&captureBuffer[FRAME_BUFSIZE / 4], ir_image_buf, ir_buf_size);

			memcpy(&captureBuffer[2 * FRAME_BUFSIZE / 4], depth_image_buf2, depth_buf_size2);
			memcpy(&captureBuffer[3 * FRAME_BUFSIZE / 4], ir_image_buf2, ir_buf_size2);
				
			LeaveCriticalSection(&CriticalSectionForCaptureCopy);
			
			try
			{
				k4a_image_release(depth_image2);
				k4a_image_release(ir_image2);
				k4a_image_release(depth_image);
				k4a_image_release(ir_image);
			}
			catch (int e)
			{
				printf("exception in image release.\n");

			}

		}
		else
		{
			printf("Ir16 or Depth None - drop frame\n");
		}
		
		
	}
	// release capture (will never be reached)
	// k4a_capture_release(capture);
	DeleteCriticalSection(&CriticalSectionForCaptureCopy);
	return 1;
}
```

Any idea what I am doing wrong ?
Thanks for reading, any help would be very much appreciated !!

Fab

RoseFlunder, 2021-03-19:
Take a look at this setting for you device that uses both sensors:
https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__device__configuration__t_a8208974f05d89fc1362c6a0900bdef4d.html#a8208974f05d89fc1362c6a0900bdef4d

fdechaumont, 2021-03-19:
Thanks RoseFlunder !

Good news is that I have no more drop
bad news is that images are still not synchronized. To get sync images, do we need to check timestamps and perform drop of images ourselves by code ?

my updated init:

```cpp
	// SUBORDINATE
	k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	config.color_resolution = K4A_COLOR_RESOLUTION_OFF;
	config.color_resolution = K4A_COLOR_RESOLUTION_720P;
	config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config.wired_sync_mode = K4A_WIRED_SYNC_MODE_SUBORDINATE;
	config.subordinate_delay_off_master_usec = 160; // Recommended min of 160uS to avoid multiple cam interference
	config.synchronized_images_only = true;

	// MASTER
	k4a_device_configuration_t config2 = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config2.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	//config.color_resolution = K4A_COLOR_RESOLUTION_OFF;
	config2.color_resolution = K4A_COLOR_RESOLUTION_720P; // Master has to grab visual image for sync purposes...
	config2.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config2.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config2.wired_sync_mode = K4A_WIRED_SYNC_MODE_MASTER;
	config2.synchronized_images_only = true;

```

RoseFlunder, 2021-03-19:
I think the green screen example might help you:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen

In that example there also two kinects used with synchronization.
It performs timestamp comparison as you already mentioned.

fdechaumont, 2021-03-19:
Thanks a lot, I will check that !

closed 2021-03-19 by fdechaumont