microsoft/Azure-Kinect-Sensor-SDK #1537: Body tracking SDK version 1.1.0 CUDA bug

Deok-Won-Lee, 2021-03-22:
Hi, I'm using Kinect Azure and the body tracking sdk.
when i excuted the "k4abt_simple_3d_viewer" there is warnning messages like below.
I think the message means that the body tracker is not using GPU. 
And it seems like there is some link error btw tracking sdk and cuda lib.

[image]

My environment is
Sensor sdk : 1.4
Body tracking sdk : 1.1
Cuda version : 11.1
cudnn version : 8.0.5
OS : Ubuntu 18.04

Help!
Thanks

qm13, 2021-03-22:
This is expected. ONNX Runtime now warns when a model instructions cannot be executed on GPU. The DirectML execution environment drops back to CPU. The CUDA execution environment drops back to CPU. The TensorRT execution environment drops back to CUDA and then CPU.

The model is still running on GPU with the exception of the Shape__40 node.

qm13, 2021-03-24:
Fixed

closed 2021-03-24 by qm13