microsoft/Azure-Kinect-Sensor-SDK #1539: model_path for body tracking model seems to not be taken in account

Moovency, 2021-03-24:
Hi, i'm trying to use the light model in our visual studio project.

I assign the value "dnn_model_2_0_lite_op11.onnx" to the model_path variable the body tracking config, but when I execute I have an error looking for the previous model "dnn_model_2_0.onnx"

[image]

I display the model_path after assigning itand the value is correct
[image]

Could you help me please ?

vladalx, 2021-03-24:
It looks like the model_path received by the TrackerHost::Create is null. Please make sure that the k4abt_tracker_configuration_t instance referenced by the GetConfigBodyTracker is the same one that is used in the CreateBodyTracker function.

Moovency, 2021-03-24:
Thank you for your quick response.
When I check the value in the create function I obtain that:
[image]

vladalx, 2021-03-24:
Are you sure you are using the right k4abt.dll from the 1.1.0 version ?

Moovency, 2021-03-24:
That was the problem, I was launching my program in a folder containing files from a previous version generation and they were taken instead of the  version.

Thank you for your help my problem is solved.

closed 2021-03-24 by Moovency