microsoft/Azure-Kinect-Sensor-SDK #1541: body track v1.1.0 inconsistent REDIST.txt, files in bin directory, and needed files documentation

diablodale, 2021-03-24:
At least three files appear inconsistent with respect to their presence (or not) in
* `sdk\windows-desktop\amd64\release\bin`
* `REDIST.txt`
* https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#required-dlls-for-onnx-runtime-execution-environments

I have so far identified the following. There may be more needed to be removed or added
```
myelin64_1.dll
nvinfer.dll
nvinfer_plugin.dll
```

### Setup

Body Tracking v1.1.0

###  Repro

Investigate the three files at the above three locations

```
myelin64_1.dll
nvinfer.dll
nvinfer_plugin.dll
```

### Result

* are not present in REDIST.txt
* are present in `sdk\windows-desktop\amd64\release\bin`
* are present in https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#required-dlls-for-onnx-runtime-execution-environments and suggests they are distributed via TensorRT 7.2.1

### Expected

To be not present in `sdk\windows-desktop\amd64\release\bin` since they are not listed in REDIST.txt.
Or to keep them in `bin` and list them in REDIST.txt
Or some other consistent intention that complies with both Microsoft and NVIDIA distribution licensing rules.

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13

diablodale, 2022-05-25:
@qm13, Fixed in what version? For customers to verify unseeable code changes, we need to know the exact version in which is was potentially fixed so that customers can verify, escalate non-fix, etc.

diablodale, 2022-05-25:
In addition... REGRESSION caused by @qm13 by commit https://github.com/MicrosoftDocs/azure-docs/commit/c8ba1e38cc2c72c3a9ae5ba80f93ad54313ab985

Docs at https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-setup have lost the file requirements for Body Tracking SDK v1.1.0
Customers are use solutions with Body Tracking v1.1.0. Documentation must continue for v1.1.0.
Just because Microsoft publishes a new SDK, that does not automagically upgrade and install customer solutions around the world. 😉
Customers buy and used fixed single version solutions, and providers sell them to customers. Providers use tools like Body Tracking SDK over years of time and therefore sale and support multiple provider produces using multiple versions of Body Tracking SDK.