microsoft/Azure-Kinect-Sensor-SDK #1542: CUDA and TensorRT runtimes missing sign op

conatsera, 2021-03-24:
**Describe the bug**
Both CUDA and TensorRT fail to load a sign op that runs on the GPU. This potentially degrades performance with the CUDA runtime as it falls back to CPU. TensorRT fails to initialize entirely and does not fallback. While the errors produced aren't exactly the same, TensorRT reports this error as a missing CUDA kernel so I believe these both are the same issue.
**To Reproduce**

**Expected behavior**

**Logs**
*CUDA*:
```
2021-03-24 10:13:42.1938108 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 onnxruntime::CUDAExecutionProvider::GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2021-03-24 10:13:42.2015767 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 onnxruntime::GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40
```
*TensorRT*:
```
2021-03-24 10:03:06.9713811 [E:onnxruntime:K4ABT_ORT_Environment, tensorrt_execution_provider.h:45 onnxruntime::TensorrtLogger::log] [2021-03-24 15:03:06   ERROR] INVALID_ARGUMENT: getPluginCreator could not find plugin Sign version 1
Unsupported ONNX data type: DOUBLE (11)
Unsupported ONNX data type: DOUBLE (11)
Unsupported ONNX data type: DOUBLE (11)
Unsupported ONNX data type: DOUBLE (11)
2021-03-24 10:03:24.6211451 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 onnxruntime::CUDAExecutionProvider::GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
```

**Desktop (please complete the following information):**
 - OS with Version: Dev 10.0.21337.1010
 - SDK Version:  1.4.1
 - Firmware version:
   - RGB Sensor Version:  1.6.110
   - Depth Sensor Version:1.6.79
   - Mic Array Version:   1.6.14
   - Sensor Config:       6109.7

**Additional context**
I have confirmed that all dependent DLLs being loaded are those that have been redistributed with the body tracking SDK. I have also tried using the latest CUDA 11.1 and TensorRT 7.2.1 binaries, but these produced the same results.

conatsera, 2021-03-24:
I missed a related issue in my search where it is expected for CUDA's sign op to run on the CPU #1537 
TensorRT stops at trying CUDA and doesn't fallback further, so I'll adjust the title

diablodale, 2021-03-24:
I am success with the TensorRT provider on my laptop RTX2070Super.

I do see similar log entries...

```
2021-03-24 17:27:18.0973665 [E:onnxruntime:K4ABT_ORT_Environment, tensorrt_execution_provider.h:45 onnxruntime::TensorrtLogger::log] [2021-03-24 16:27:18   ERROR] INVALID_ARGUMENT: getPluginCreator could not find plugin Sign version 1
(4) Unsupported ONNX data type: DOUBLE (11)
Loaded 'C:\njs\dp.kinect3\build\cudnn_ops_infer64_8.dll'. Module was built without symbols.
Loaded 'C:\njs\dp.kinect3\build\nvrtc-builtins64_111.dll'. Module was built without symbols.
2021-03-24 17:27:27.4356607 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 onnxruntime::CUDAExecutionProvider::GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
Loaded 'C:\njs\dp.kinect3\build\cudnn_cnn_infer64_8.dll'. Module was built without symbols.
```

However, if I wait ~35 seconds, then I can see 24 files slowly appear in the Tensor Cache directory. And soon after I start receiving good body tracking frames.

conatsera, 2021-03-24:
Hmm, in my use case I run this while in VR where CPU usage is at constant 100% even after loading any of the models. I had waited about 2 minutes but maybe that wasn't enough. I will let it sit for 5 minutes and see what happens

conatsera, 2021-03-25:
Looks like it just takes around 3-4 minutes to load. I'm changing runtimes while in steamvr so my poor CPU just can't handle loading it in a reasonable time. I was hoping for something a little faster, but I'll just have to launch TensorRT first when trying a side-by-side comparison.

closed 2021-03-25 by conatsera

diablodale, 2021-03-25:
FYI other readers, if I use TensorRT with the fp16 option, my startup time greatly increases; from ~35 seconds to 4+ minutes. Yet after that, it runs with no errors. And if TensorRT caching is used, the next startup time is only seconds.