microsoft/Azure-Kinect-Sensor-SDK #1543: New BodyTracking not working in Unity

hvallee, 2021-03-24:
Hi,

We successfluy made the new version of the body tracker work in our C++ visual studio project. We use this project to generate a DLL for Unity and it has been working so far.
But we can't make this new version work. It crashes at the body tracker creation.
There are plenty of new dll with this version and I imagine some must be at the root of the unity project and some in the plugin folder. 
But I had no success for now.
Has anyone some success in the atempt ?

Thank you for your help;

qm13, 2021-03-25:
Is the Unity DLL based on https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking

qm13, 2021-03-25:
You need to make sure the new DLLs are in the correct location for Unity

Vangos, 2021-03-30:
I have a solution -- it's mostly a workaround. You can check out LightBuzz's [Unity Asset Store plugin](http://u3d.as/1FBs) that automates the following process for you. Otherwise, read on:

The problem has to do with the **location** of the DLL files. We'll see how to do this in the Editor and the executable Build.

## 1) Binaries

First, you need to collect the required binary files. Here's where to find them:

- Download and install the [Azure Kinect Sensor SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download) for Windows.
- Download and install the [Azure Kinect Body Tracking SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download) for Windows.

The required DLL binaries are located in the following folders:

- `C:\Program Files\Azure Kinect Body Tracking SDK\tools`
- `C:\Program Files\Azure Kinect Body Tracking SDK\sdk\netstandard2.0\release`
- `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\netstandard2.0\release`

Copy all of the `.dll` and `.onnx` files from the above locations and move to the next step. You should have 21 `.dll` files and 2 `.onnx` files.

## 2) Unity Editor

To properly run Kinect 4 Azure body-tracking in the Editor, you need to copy the above DLLs **next to your Assets** project folder, like this:

[image]

The `.onnx` model files should also be there.

**Attention**: Do not place the above binaries in your `Plugins/x86_64` folder!

## 3) Unity installation

That's the tricky part. You should manually copy the `DirectML.dll` file into the installation folder of Unity. In my case, it is:
`C:\Program Files\Unity\Hub\Editor\2020.3.0f1\Editor`.

[image]

It's recommended to reboot Unity3D and restart your computer after this step.

## 4) Building your project

The above method will only work in the Editor. After you export your program, you also need to copy the DLL files **next to your `.exe` file**.

[image]

That's it. Unity Editor and Unity builds should work fine now 😄 

_Tested with Unity 2019 LTS and 2020 LTS._

Again, you can download the LightBuzz [Unity Asset Store plugin](http://u3d.as/1FBs) that automates the above process for you. My apologies for the shameful product placement 😉

closed 2021-04-14 by hvallee