microsoft/Azure-Kinect-Sensor-SDK #1544: Camera and IMU connection problems

Simonalexanderadams, 2021-03-24:
**Describe the bug**
Azure Kinect was connecting okay this morning for depth streams, but had issues also getting color steams. I noticed the Kinect needed a firmware update and updated. After the firmware update the device refused to connect to any camera or IMU, though the microphone works consistently. 

The following messages pop up when connecting:
"IMU Failed: Timeout"
"000135594912: IMU Data: data source failed"
"Cameras failed: timed out!"

**Troubleshooting attempted thus far**
- I have tried this  Kinect on three computers, for which this device and other Azures have worked fine in the past. I get similar issues for each, logs below.
- I have tried multiple ports for each computer, and multiple cables
- I have tried only opening on camera, and on different resolution and FPS settings
- I've closed other software tasks that might be accessing the camera
- As far as I can tell the Kinect enumerates correctly in device manager. As noted, all USB Hosts have worked in the past. 
[image]
- I did a factory reset, and the problem was the same. Continued after a 2nd update to firmware. Firmware changes are noted correctly in the Kinect Viewer 1.4.1
- Unfortunately I do not have access to a 2nd Azure at this time (would help root out computer vs. device issue), though testing three separate computers that are known to work with Azure in the past does suggest to me it is a device issue.

**Logs**
Computer A and C:
[ warning  ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbstreaming.c:82 ) : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
_(repeated multiple times)_
[ error    ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbstreaming.c:117 ) : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected
_(repeated multiple times)_

Computer B (provided an additional error):
[ warning  ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbstreaming.c:82 ) : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
_(repeated multiple times)_
[ error    ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbstreaming.c:117 ) : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected
_(repeated multiple times)_
[ error    ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\mfcamerareader.cpp:1175 ) : OnFlush(). Failed to deselect stream for stop: 0x800706bf

**Desktop (please complete the following information):**
- Computer A and B: Windows 20H2, Build 19042.867
- Computer C: Windows 1909, Build 18363.1440
 - SDK Version:  1.4.1
 - Firmware version: 
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

I've reached the end of what I can do with sorting through old bug reports, and troubleshooting guides. Hopefully someone here has an answer. Happy to provide more info as requested.

qm13, 2021-03-25:
@Simonalexanderadams it sounds like your Azure Kinect DK has died. Please follow [these ](https://support.serviceshub.microsoft.com/supportforbusiness/create?sapId=f77b1b95-721e-43a0-2db8-b01e81a3f813) instructions to get it replaced.

Simonalexanderadams, 2021-03-25:
> @Simonalexanderadams it sounds like your Azure Kinect DK has died. Please follow [these ](https://support.serviceshub.microsoft.com/supportforbusiness/create?sapId=f77b1b95-721e-43a0-2db8-b01e81a3f813) instructions to get it replaced.

Really? There's no way to recover it? I'm surprised that the only way around this is to ask Microsoft for a new one... I have plenty of version 2 Kinects that are still functional after years of no support, while this Azure which is relatively new in comparison just bit the bullet out of the blue.

I'd be interested in more of an explanation as to why you've concluded that it's dead, but if it is indeed bricked I can go through that process from your link.

Simonalexanderadams, 2021-03-26:
Update:
I plugged the device in again to double check and it worked fully with the following warning: 
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 36ms (this may be transient).

I switched to TouchDesigner (my use case) and was getting good readings on depth and point cloud at the same time, but they both failed when I opened the RGB steam (usually it is fine to grab all three.)

I tried switching USB ports but get this error when ejecting it in safely remove hardware:
[image]
I still switched ports (probably a mistake) and now it is giving me the same errors as my initial post again.

qm13, 2021-03-26:
@Simonalexanderadams we have seen very few (less than Kinect for Windows by volume) failures of Azure Kinect DKs. You have gone through all the debug steps we would suggest hence my conclusion. That said it sounds like you have multiple Azure Kinects. Can you verify that one of the other devices works fine with the PCs. This will eliminate the PCs. Failing that can you try webcam software (test RGB and mics only) with both AKDK and a USB webcam.

The [ warning ] is expected. It is means the depth engine tool longer than the frame interval to process the raw depth frames. Usually occurs when there is not enough GPU head room to cover depth engine, Windows and applications running.

zaalsabb, 2021-04-25:
Can you reopen this issue?

qm13, 2021-04-26:
@zaalsabb please describe the symptoms you are seeing.

zaalsabb, 2021-04-27:
@qm13 While I was using the Azure Kinect for streaming data, it suddenly stopped sending images, depth, and imu data. The issue still persists even after rebooting the computer and even when I use a different computer and different USB cables. I also encountered this issue on both Windows 10 and Ubuntu 18.04.

When I open k4aviewer I see the following errors:

(repeated) [ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
(repeated) [ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected

and also:
"IMU Failed: Timeout"
"Cameras failed: timed out!"

I am using the latest SDK 1.4.1. 
I also tried doing factory reset as suggested here, but the issue still persists.

qm13, 2021-04-27:
@zaalsabb it really sounds like your Azure Kinect has died. Do you have another AKDK that you can use to verify the computer setup and USB cable are ok?

qm13, 2021-04-27:
Also does the device enumerate in device manager?

zaalsabb, 2021-04-27:
Unfortunately I don't have another Azure Kinect. I tried it on another computer with another USB cable and the problem still persists. The device does enumerate correctly in the device manager though.

qm13, 2021-04-27:
Can you access the RGB camera and mic array from another Windows application e.g. webcam.

zaalsabb, 2021-04-29:
I tried to access the RGB camera using Zoom since the device is listed as a webcam option. However, when I select it all I get is a black screen. As for the mic, Zoom is able to receive the signal with no issue, as seen in the bottom left corner of the second image I attached.

[image]
[image]

qm13, 2021-04-29:
I assume you see the same problem with the RGB camera on other Windows systems. It really sounds like you have had a partial failure of your Azure Kinect DK. You can find instructions to replace the device [here](https://support.serviceshub.microsoft.com/supportforbusiness/create?sapId=f77b1b95-721e-43a0-2db8-b01e81a3f813).

UnityQueriesAsk, 2021-08-19:
Can you reopen this issue @qm13? I am also facing the same issue with Azure Kinect DK. In my case I am getting the following messages pop up when connecting:

IMU failed: timed out!
IMU Data: data source failed!
Camera failed : timed out!

I am attaching my USB host controllers and the errors that I am getting in the Azure Kinect Viewer, Please check :

USB host controllers :
https://drive.google.com/file/d/14X5jzVEvOc-qkmrb0CJ-Y0cuWMEBWUFL/view
https://drive.google.com/file/d/1XPeXY5QC_o9fDfEqPZ07C6EE7Rwz5FXK/view

Errors from Azure Kinect Viewer :
https://drive.google.com/file/d/1Iu6VrUK06LyiLto0Yy_9dSzyNyzEWwz6/view

I have been using the Azure Kinect DK device for long days but suddenly 2 days back it is failed to open the Color & Depth & Infrared cameras in the Azure Kinect Viewer.

I have checked the Kinect device with another windows camera application in that case the camera is working fine but the Kinect device cameras are failed to open in the Azure Kinect Viewer.

Can you please provide a solution for these issue.

Thank you.

qm13, 2021-08-19:
Have you tried the AKDK on another computer?

qm13, 2021-08-19:
Reclosing. Covered by #1666.

closed 2021-08-19 by qm13

TheUnproductive, 2022-03-20:
I don't know if it is relevant anymore, but I managed to get the Kinect working again after reinstalling the device