microsoft/Azure-Kinect-Sensor-SDK #1546: body track v1.1.0 with DirectML -- slower than v1.1.0 CUDA and v1.0.1 GPU

diablodale, 2021-03-24:
I am unclear if I should be pleased, disappointed, or if there is a functional bug regarding the relative poor performance of the DirectML onnx provider on an NVidia RTX2070Super. I acknowledge I see nothing written with a promise DirectML would perform like CUDA.

The DirectML provider appears:

1. CPU bound
2. lag of ~800ms
3. can not sustain 30fps

This is in contrast to the CUDA onnx provider. And in contrast to the original GPU provider for BT v1.0.1 (which was itself CUDA).

### Setup

* Microsoft Windows [Version 10.0.19042.870]
* VS2019 Community v16.9.2 building 64-bit
* Sensor SDK 1.4.1
* Body Track SDK 1.1.0
* App of your choice; I used my own dp.kinect3 app
* Intel i7-10875H (8 core, 16 threads, 2.3->5GHz) with 32GB ram and RTX2070Super

### Repro

1. Setup and compile a release build as above
2. standard onnx model file
3. Output depth/ir/bodyindex at 640x576 , color nv12 at 1280x720, and skeleton joints

### Result

compute | lag (ms) | fps | cpu
-- | -- | -- | --
CPU | 4000 | 4.5 | bound; avg=70%
DirectML | 800 | 23 | bound; avg=83%
CUDA | < 200 | 30 | not bound; avg=30%
Tensor | < 200 | 30 | not bound; avg=28%
v1.0.1 GPU | < 200 | 30 | not bound; avg=27%

### Expected

It is more a hope...that DirectML would be competitive with CUDA. Its current performance results in high lag and overconsumption of CPU resources that results in other components of solutions not able to do either own computations in a timely manner.

I acknowledge I see nothing written with a promise DirectML would perform like CUDA. I also acknowledge that DirectML does appear to work (slowly) on other manufacturer GPUs like Intel integrated GPUs. So...the story is improving 🙂

The DirectML provider will not work in parallel -- as compared to the CUDA provider which can work in parallel. I am sending capture frames asap to the BT API so if it can't accept in parallel, then this might contribute to the lower fps and higher lag. What is the Microsoft teams perspective on this? https://www.onnxruntime.ai/docs/reference/execution-providers/DirectML-ExecutionProvider.html#multi-threading-and-supported-session-options

### Graphs

Memory usage on all was steady. I saw no obvious memory leaks.
CUDA and TensorRT have a curious ramp that cycles forever from ~65% -> 100%, stays at 100% for many seconds, then back to the 65->100 ramp. This same cycle exists on BT v1.0.1. And across NVidia drivers from Dec 2020->March 2021 both GameReady and Studio. 30 fps is maintained throughout, though, there may or may not be remaining GPU for other tasks.

#### DirectML

CPU cores are chaotic; one is completely saturated which is likely something single-threaded
Both GPUs are not well utilized, the graphs suggest only 28% used.

[image]

#### CUDA

CPU usage is more steady. None are saturated or bound.
The NVidia GPU is very used.

[image]

#### TensorRT

Almost identical to CUDA

[image]

#### Old BT v1.0.1 GPU

For comparison, this is the old BodyTrack v1.0.1 GPU. It is also very similar to the newer "CUDA"

[image]

diablodale, 2021-03-25:
My gut tells me there is a bug in my or the BT SDK's code. Though it is seen only with the DirectML backend and I don't change my code except passing the DIRECTML enum. When I look at the behavior and more data today, it is as if part of Body Tracking v1.1.0 is falling back to CPU and not using the GPU path. I'm seeing the CPU saturated and the GPUs barely used.

My app is multithreaded. And I make use of Intel TBB. Due to this, there is not a single consistent thread which calls the BT apis. It is nothing like the single-threaded and sequential `simple_3d_viewer.exe`. On my 8core/16 thread CPU, there could be 16 threads simultaneously calling BT apis like `enqueue_capture()` and `pop_result()`. It is not sequential; meaning there is no guarantee thread 42 calls enqueue() and after that same thread 42 calls pop(). Instead, thread 721 might call pop() and thread 22 might call enqueue(). All can happen in any order.

In addition, it is not guaranteed *which* threads make the BT api calls. It could be the same set of 16 threads, or it could be a constantly changing set of threads. I highlight this because if you are using TLS (thread local storage), then you could have a bug related to such usage. You might store something important in TLS and then I or TBB destroys that thread and it is never seen again.

diablodale, 2021-03-27:
Also consider that `k4a.dll` and `k4abt.dll` is delay loaded.

Rough sequence:

1. The main App.EXE is a third-party. A customer clicks the app's icon in start menu.
2. Sometime during the running of the app, the customer chooses to enable my Kinect "plugin" DLL that I code/sell.
3. My DLL runs and does things without any active Kinect sensor activity.
4. Customer chooses "go!".
5. My DLL opens the Kinect device (loads `k4a.dll`), opens camera, create BodyTracker (loads `k4abt.dll)`, etc.

It is impossible to load Azure Kinect DLLs on startup in the main app.exe. I am not the 3rd party.
It is difficult and degrades my product to load Azure Kinect DLLs immediately with my "plugin" DLL.
Both of these and the 2.5 GB of Body Tracking DLLs/models (that many customers don't need) drive my use of the delay-load approach.

There are material differences when DLLs are delay loaded (e.g. TLS). I recommend your team consider if any Kinect code is written in a restrictive way that breaks when delay-loaded. Docs at https://docs.microsoft.com/en-us/cpp/build/reference/linker-support-for-delay-loaded-dlls?view=msvc-160#constraints-on-delay-load-dlls

diablodale, 2021-03-28:
I've made repro code and can reproduce similar results as in OP.

### Repro code setup

1. Same setup and hardware as in OP
2. Visual Studio as in OP with Windows SDK 10.0.19041.0
3. git clone https://github.com/diablodale/kinectazure-bug-1546
4. Set some env vars so it can find Kinect SDKs, or edit cmake to hard code them
4. cmake config with Windows SDK 10.0.19041.0 and 64-bit compile
5. cmake build release
6. Run the EXE with various parameters. Usage is documented in the code or running the EXE with no parameters.

### Results

exe params | fps | cpu usage | Intel gpu usage | Nvidia gpu usage | Nvidia gpu temp | avg Nvidia power (115w max) | Nvidia temp throttle
-- | -- | -- | -- | -- | -- | -- | --
0 directml | 2.6 | 15% steady | 93% steady | 0% | 54c | 15w | no
1 directml | 24.1 | 85% chaotic and 2+ bound | 18% steady | 2-49% chaotic | 70c | 104w | no
1 directml 16 | 24.2 | 84% bound and chaotic | 18% steady | 2-39% chaotic | 75c | 110w | no
1 directml 33 | 23.8 | 80% bound and chaotic | 17% steady | 2-34% chaotic | 74c | 104w | no
0 cuda | 30 | 18% steady and 1 bound | 17% steady | 0-100% see-saw pattern | 74c | 107w | no
0 cuda 16 | 29.9 | 16% steady and 1 bound | 18% steady | 0-100% see-saw pattern | 73c | 104w | no

#### 0 directml

[image]

#### 1 directml

[image]

#### 1 directml 16

[image]

#### 1 directml 33

[image]

#### 0 cuda

The gpu index for CUDA does not match the index used for DirectML 🙃

[image]

#### 0 cuda 16

[image]

senshin998, 2021-05-30:
I have the same doubts，  poor performance for  body track v1.1.0 ,why ?

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13

diablodale, 2022-05-25:
@qm13, Fixed in what version? For customers to verify unseeable code changes, we need to know the exact version in which is was potentially fixed so that customers can verify, escalate non-fix, etc.