microsoft/Azure-Kinect-Sensor-SDK #1547: Body tracking performance is too low in C# winform.

ykk271, 2021-03-25:
I made a simple example to save RGB video and skeleton data. but FPS is too low. When capturing, CPU and GPU use is only 20% and 10% respectively. How to improve FPS or GPU usage rate? The FPS is calculated between 15 and 10.

**Setup**
- Sensor SDK 1.4.1
- Body tracking SDK 1.0.1
- CPU: AMG Ryzen 7 3700X
- GPU: RTX2060

qm13, 2021-03-25:
Can you share the simple example?

ykk271, 2021-03-26:
Thank you for your response.

My code here. 
===============================================
using System;
using System.Drawing;
using System.Windows.Forms;
using Microsoft.Azure.Kinect.Sensor;
using Microsoft.Azure.Kinect.BodyTracking;
using Image = Microsoft.Azure.Kinect.Sensor.Image;
using PixelFormat = System.Drawing.Imaging.PixelFormat;
using System.Buffers;
using System.Threading;

namespace Gait_Kinect
{
    public partial class Form1 : Form
    {
        Device Kinect;
        Tracker bodyTracker;
        Calibration calib;
        Skeleton skl;

        Bitmap ColorBitmap;
        bool Loop = true;

        Thread t;
        Thread t2;

        public Form1()
        {
            InitializeComponent();
            InitKinect();

            t = new Thread(new ThreadStart(KinectLoop));
            t.Start();

            t2 = new Thread(new ThreadStart(KinectLoop2));
            t2.Start();
        }

        private void KinectLoop()
        {
            while (Loop)
            {
                unsafe
                {
                    Image colorImage = Kinect.GetCapture().Color;
                    using (MemoryHandle pin = colorImage.Memory.Pin())
                    {
                        ColorBitmap = new Bitmap(
                             colorImage.WidthPixels,
                             colorImage.HeightPixels,
                             colorImage.StrideBytes,
                           PixelFormat.Format32bppArgb,
                             (IntPtr)pin.Pointer);
                    }
                }
                PictureBox_RGB.Image = ColorBitmap;
            }
        }

        private void KinectLoop2()
        {
            while (Loop)
            {   
                bodyTracker.EnqueueCapture(Kinect.GetCapture());
                Frame frame = bodyTracker.PopResult();
                if (frame != null)
                {
                    uint bodyNum = frame.NumberOfBodies;
                    if (bodyNum > 0)
                    {
                        skl = frame.GetBodySkeleton(0);
                    }
                }
                frame.Dispose();
            }
        }

        private void InitKinect()
        {
            Kinect = Device.Open(0);

            Kinect.StartCameras(new DeviceConfiguration
            {
                ColorFormat = ImageFormat.ColorBGRA32,
                ColorResolution = ColorResolution.R1536p,
                DepthMode = DepthMode.NFOV_2x2Binned,
                SynchronizedImagesOnly = true,
                CameraFPS = FPS.FPS30,
            });

            calib = Kinect.GetCalibration();
            bodyTracker = Tracker.Create(calib, new TrackerConfiguration
            {
                ProcessingMode = TrackerProcessingMode.Gpu,
                SensorOrientation = SensorOrientation.Default,
                GpuDeviceId = 1
            });
        }

        private void Form1_FormClosing(object sender, FormClosingEventArgs e)
        {
            Loop = false;
            Kinect.StopCameras();
            t.Abort();
            t2.Abort();
        }
    }
}

ykk271, 2021-03-26:
Thanks for your response.
my code is here.

```csharp
using System;
using System.Drawing;
using System.Windows.Forms;

using Microsoft.Azure.Kinect.Sensor;
using Microsoft.Azure.Kinect.BodyTracking;

using Image = Microsoft.Azure.Kinect.Sensor.Image;
using PixelFormat = System.Drawing.Imaging.PixelFormat;
using System.Buffers;
using System.Threading;

namespace Gait_Kinect
{
    public partial class Form1 : Form
    {
        Device Kinect;
        Tracker bodyTracker;
        Calibration calib;
        Skeleton skl;

        Bitmap ColorBitmap;
        bool Loop = true;

        Thread t;
        Thread t2;

        public Form1()
        {
            InitializeComponent();
            InitKinect();

            t = new Thread(new ThreadStart(KinectLoop));
            t.Start();

            t2 = new Thread(new ThreadStart(KinectLoop2));
            t2.Start();
        }

        private void KinectLoop()
        {
            while (Loop)
            {
                unsafe
                {
                    Image colorImage = Kinect.GetCapture().Color;
                    using (MemoryHandle pin = colorImage.Memory.Pin())
                    {
                        ColorBitmap = new Bitmap(
                             colorImage.WidthPixels,
                             colorImage.HeightPixels,
                             colorImage.StrideBytes,
                           PixelFormat.Format32bppArgb,
                             (IntPtr)pin.Pointer);
                    }
                }
                PictureBox_RGB.Image = ColorBitmap;
            }
        }

        private void KinectLoop2()
        {
            while (Loop)
            {   
                bodyTracker.EnqueueCapture(Kinect.GetCapture());
                Frame frame = bodyTracker.PopResult();
                if (frame != null)
                {
                    uint bodyNum = frame.NumberOfBodies;
                    if (bodyNum > 0)
                    {
                        skl = frame.GetBodySkeleton(0);
                    }
                }
                frame.Dispose();
            }
        }

        private void InitKinect()
        {
            Kinect = Device.Open(0);

            Kinect.StartCameras(new DeviceConfiguration
            {
                ColorFormat = ImageFormat.ColorBGRA32,
                ColorResolution = ColorResolution.R1536p,
                DepthMode = DepthMode.NFOV_2x2Binned,
                SynchronizedImagesOnly = true,
                CameraFPS = FPS.FPS30,
            });

            calib = Kinect.GetCalibration();
            bodyTracker = Tracker.Create(calib, new TrackerConfiguration
            {
                ProcessingMode = TrackerProcessingMode.Gpu,
                SensorOrientation = SensorOrientation.Default,
                GpuDeviceId = 1
            });
        }

        private void Form1_FormClosing(object sender, FormClosingEventArgs e)
        {
            Loop = false;
            Kinect.StopCameras();
            t.Abort();
            t2.Abort();
        }
    }
}

```

YJ-L-2017, 2022-03-02:
I have the same problem,CPU and GPU use is only 50% and 20% respectively,The FPS is calculated between 15 and 30.
Setup

Sensor SDK 1.4.1
Body tracking SDK 1.0.1
CPU: i9 9880
GPU: RTX1650

fkavache, 2022-04-18:
Hi @YJ-L-2017,

Could you please provide a simple example project to reproduce this issue?

closed 2022-05-17 by qm13

Thaina, 2022-10-26:
In unity also slow, around 100 ms per `GetCapture`. And so the latency are very obvious. Unlike `Azure Kinect Body Tracking Viewer` 3D visualization app that was very fast

Is it because I use `TrackerProcessingMode.Cuda` ?