microsoft/Azure-Kinect-Sensor-SDK #1548: How to get the distance information of a pixel in a depth image?

zqz979666, 2021-03-25:
as the title. New to azure kinect.
I got the depth image of a capture. However, I can only derive the grayscale value of a certain pixel, can't get the specific distance between it and the sensor. Is there any algorithm to calculate or sth else? Thx!

qm13, 2021-03-25:
The pixel value of the depth image is mm from the camera. I suggest you look at the k4aviewer https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/tools/k4aviewer. The colored depth view shows how to convert the depth values to color range. If you run k4aviewer and hover the mouse over a pixel the viewer reports the distance of the pixel from the camera.

zqz979666, 2021-03-26:
> The pixel value of the depth image is mm from the camera. I suggest you look at the k4aviewer https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/tools/k4aviewer. The colored depth [...]

Thanks for answering! I think I have understood.

closed 2021-03-26 by zqz979666