microsoft/Azure-Kinect-Sensor-SDK #1550: How to include Body Tracking SDK without using NuGet

Kevin0811, 2021-03-27:
I am working on a project on Visual Studio
In previous version I include the sdk by following the Azure Kinect DK DOCs using NuGet
But in this version the change log tells that "[Breaking Change] No NuGet install."
It seems that previous method doesn't work anymore? 
but also add "[Feature] Verified samples compile with Visual Studio 2019 and updates samples to use this release."
However I have no idea how it work, maybe this is not even a problem to others, but I'm a junior developers without many experiences.  Looking forward to hear the solution from request, it will help a lot, thanks you very much.

MarkYu98, 2021-04-01:
Same here, have you found a solution? Thx.

Kevin0811, 2021-04-01:
Yes, Thank you

MarkYu98, 2021-04-01:
Hi @Kevin0811, can you share how you managed to make it work? I have installed the Body Tracking API 1.1.0 seperately and add the corresponding headers, libraries, and DLLs to my VS2017 project. I can build and run the simple_3d_viewer example but the fps seems much smaller than the provided simple_3d_viewer.exe.

closed 2021-04-13 by qm13

fractalfantasy, 2021-04-22:
if anyone has insight on how to install without NuGet that would be much appreciated

MarkYu98, 2021-04-23:
> if anyone has insight on how to install without NuGet that would be much appreciated

Yes I have figured it out. It's pretty simple actually. Just download the sdk and unzip it. Then in your VS project's property settings, add the "include" path to C/C++->Additional Include Path; add the "lib" path to Linker->Addtional Library Path; add "k4abt.lib" (along with "k4a.lib", "k4arecord.lib" if needed) to Linker->Input->Additional Dependencies. Then it should be OK to run.