microsoft/Azure-Kinect-Sensor-SDK #1555: [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign

xhniu, 2021-04-01:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**
➜  lib64 sudo k4abt_simple_3d_viewer
2021-04-01 17:30:58.076544763 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2021-04-01 17:30:58.078060216 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: ubuntu 18.04
 - SDK Version:  1.4.1
 - Firmware version: 
 - RGB Sensor Version: 1.6.110
 - Depth Sensor Version:1.6.79
 - GPU: Nvidia GTX 3070
 - GPU Drivers tested: 455.45.01

**Additional context**

diablodale, 2021-04-01:
known and team writes it should be ignored https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#specifying-onnx-runtime-execution-environment

closed 2021-04-13 by qm13

bearpaw, 2021-11-03:
I got the same warnings with v1.1 and it led to a crash
```bash
$ k4abt_simple_3d_viewer 
2021-11-03 10:08:12.913785632 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2021-11-03 10:08:12.916377666 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40
*** buffer overflow detected ***: k4abt_simple_3d_viewer terminated
```
Seems [this issue](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/188) is related? Could anybody take a look into this? Thanks!

haeginh, 2021-11-16:
same here
*** buffer overflow detected ***: k4abt_simple_3d_viewer terminated
shows up.

haeginh, 2021-11-17:
> I got the same warnings with v1.1 and it led to a crash
> 
> ```shell
> $ k4abt_simple_3d_viewer 
> 2021-11-03 10:08:12.913785632 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
> 2021-11-03 10:08:12.916377666 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40
> *** buffer overflow detected ***: k4abt_simple_3d_viewer terminated
> ```
> 
> Seems [this issue](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/188) is related? Could anybody take a look into this? Thanks!

Hey, I solved this!
I think this seems happen when you use the latest nvidia-driver.
Please remove your current nvidia-driver and install nvidia-driver-465, reboot and try again!
This worked for me :-)

LU-MING-HUA, 2021-11-18:
> > 我在 v1.1 中收到了相同的警告，並導致了崩潰
> > ```shell
> > $ k4abt_simple_3d_viewer 
> > 2021年11月3日10：08：12.913785632 [W：onnxruntime：K4ABT_ORT_Environment，cuda_execution_provider.cc:1885 GetCapability] CUDA內核沒有發現在登記的作品類型：註冊節點名稱：註冊
> > 2021年11月3日10:08： 12.916377666 [W：onnxruntime：K4ABT_ORT_Environment，fallback_cpu_capability.h：140 GetCpuPreferedNodes]強制回退到CPU執行用於節點：Shape__40
> >  ***緩衝區溢出檢測***：k4abt_simple_3d_viewer終止
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 似乎[這個問題](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/188)是相關的？有人可以看看這個嗎？謝謝！
> 
> 嘿，我解決了這個問題！ 我認為當您使用最新的 nvidia 驅動程序時，這似乎會發生。 請刪除您當前的 nvidia-driver 並安裝 nvidia-driver-465，重新啟動並重試！ 這對我有用:-)

I have seen a lot of 465 versions. Is there a more accurate confirmation of which version it is?

haeginh, 2021-11-18:
> I have seen a lot of 465 versions. Is there a more accurate confirmation of which version it is?

In my case, it's 465.19.01 :-)

LU-MING-HUA, 2021-11-18:
I tried a lot of drivers in my case
My GPU is RTX3080
--------------DCH version------------------
466.77
472.12
--------------Standard version-------------
471.41
471.68
471.96
472.12
Can support me
The 472.12 version is the highest

fkavache, 2022-04-11:
Hi, @LU-MING-HUA

could you please take a look at this thread?

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984