microsoft/Azure-Kinect-Sensor-SDK #1561: Enumeration Errors on Ubuntu 18.04

H-Park, 2021-04-02:
Because Issue #375 , and #1260 are closed, I am opening this issue to inform the developers that the Azure Kinect is NOT stable on Ubuntu 18.04. 

I have two desktops running 18.04, and each are experiencing issues around enumerating the azure kinect.

Machine 1:
lsusb
`Bus 001 Device 019: ID 045e:097e Microsoft Corp. 
Bus 001 Device 020: ID 045e:097c Microsoft Corp. 
Bus 001 Device 021: ID 045e:097d Microsoft Corp. 
Bus 001 Device 018: ID 045e:097b Microsoft Corp. 4-Port USB 2.0 Hub`

ID 045e:097a is missing.

dmesg
`[    7.184468] uvcvideo: Found UVC 1.00 device Azure Kinect 4K Camera (045e:097d)
[    7.184601] uvcvideo: Failed to query (GET_INFO) UVC control 3 on unit 1: -32 (exp. 1).
[    7.185517] uvcvideo: Failed to set UVC probe control : -32 (exp. 26).
`

Machine 2:

lsusb:
`Bus 001 Device 019: ID 045e:097e Microsoft Corp. 
Bus 001 Device 020: ID 045e:097a Microsoft Corp. 
Bus 001 Device 021: ID 045e:097d Microsoft Corp. 
Bus 001 Device 018: ID 045e:097b Microsoft Corp. 4-Port USB 2.0 Hub`

ID 045e:097c is missing.

dmesg
`device not responding to setup address
device not accepting address 7, error -71`

The first Kinect launched 11 years ago. Since day one of this venture, Microsoft has neglected Linux, ranging from outright refusing to support it and forcing a third party developer to write an [interface](https://github.com/OpenKinect/libfreenect) for it, to refusing to consider investigating current issues being report with this model.

qm13, 2021-04-13:
@H-Park I fail to see the relevance of #375 and #1260 to your issue. They seem unrelated.

We are absolutely behind Linux and expect it to overtake Windows as the preferred OS for Azure Kinect. What is the USB controller brand you are attaching the Azure Kinect via? Please see https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#usb3-host-controller-compatibility. Assuming you are using a compatible controller can you please provide the log files.

closed 2021-04-26 by qm13

H-Park, 2021-05-05:
Sorry, we were exploring other hardware options for our product. Here are some logs.

```
[error] [t=1262] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[error] [t=1262] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[error] [t=1262] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size)
[error] [t=1001] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[error] [t=1001] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (451): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[error] [t=1001] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[error] [t=1001] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
```