microsoft/Azure-Kinect-Sensor-SDK #1562: k4abt_simple_3d_viewer source?

fractalfantasy, 2021-04-07:
We’re having trouble getting Body Tracking SDK (1.1.1) running with its new environment variables. Getting no errors / exceptions / crashes, it just doesn't seem to start. (gyp recompilation is succeeding, and CPU mode works).

Would be helpful if you updated the samples repository so that we can have a look at the k4abt_simple_3d_viewer source.

qm13, 2021-04-08:
The environment variables are read by ONNX Runtime (DLL). There is no changes needed to the samples.

What operating system are you using? What environment variables are you using?

mikkamikka, 2021-04-08:
@qm13 We are using Windows 10 (v.1909)
Environment variables we've set at Windows system user level:
```
ORT_TENSORRT_ENGINE_CACHE_ENABLE=1
ORT_TENSORRT_CACHE_PATH="C:\Temp"
```
Are the any special 'ONNX Runtime environment variables'??

Tracker configuration set to default:
`k4abt_tracker_configuration_t tracker_config = K4ABT_TRACKER_CONFIG_DEFAULT;`
I guess those Tensor related env. variables makes no sense since we using default processing mode K4ABT_TRACKER_PROCESSING_MODE_GPU_DIRECTML
(https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.1.x/k4abttypes_8h_source.html)

Another question is: 
Should we have CUDA 11.1+ and CuDNN v.8+ installed in the system to run body tracking v1.1.0 ?

qm13, 2021-04-13:
I am confused. Are you attempting to use CUDA, DirectML, or TensorRT?

All the necessary NVIDIA files are included in the .msi install. They do not need to be system wide. You can drop them into your project folder. You need the correct NVIDIA driver for your card. See troubleshooting section of the documentation.

mikkamikka, 2021-04-13:
@qm13 We are attempting to use DEFAULT processing mode: DirectML

GPU driver is up to date: NVIDIA driver version 465.89

Logging shows that dependencies check, ONNX runtime, and tracker creation is successful:
```
New logging session started

[2021-04-12 20:21:37.199] [info] [t=14612] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (465): VerifyDependencies(). Verifying Dependencies!
[2021-04-12 20:21:37.306] [info] [t=14612] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (146): Create(). Initializing CUDA and ONNX Runtime... 
[2021-04-12 20:21:54.711] [info] [t=14612] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (168): Create(). Done.
```

I've noticed that the k4a.dll you deliver with the Body Tracking SDK v1.1.0 example (on the left) differs in size from the one delivered with the Kinect SDK v1.4.1 (on the right), while the file version is the same:
[image]

Maybe this is the reason for the problem?

scotthsu98052, 2021-04-14:
@vladalx please check with fractalfantasy about the directML issue.

fractalfantasy, 2021-04-22:
Perhaps we're installing it wrong? 

With NuGet now broken on the latest update, it's difficult to understand how to update the Body Tracking SDK on the **k4abt_simple_3d_viewer** sample. We're simply replacing files within BT SDK 1.0.1 package folder - I admit this is a hacky solution, curious how you envisioned us to use this BT release without NuGet?

Here's the steps we're taking:
1) install the NuGet packages, and build solution (vs2019)
[image]

2) replace these files with the ones from the new BT SDK 1.1.0
[image]
[image]
[image]

3) copy these DLL's into the bin/release folder
[image]

Thanks for looking into this!

vladalx, 2021-04-23:
Hello

pls clarify the steps you are taking to get to this error, and let me know :

1. if you run k4abt_simple_3d_viewer from the Program Files/azure kinect bodytracking sdk/tools folder, is it working ?
2. are you using the simple_3d_viewer sample source code from git ?

Thank you

fractalfantasy, 2021-04-29:
Thanks @vladalx 

1. Yes  k4abt_simple_3d_viewer.exe from Program Files works
2. Yes we are using the sample code from git

--

We actually got the sample running by replacing .dlls in the v1.0.1 NuGet but there were stability issues (it was a hacky solution). 

Decided to take another route using a third party node, everything seems installed fine but getting this error when trying to initiate the tracker. 

Does this error tell you anything?  Perhaps it's not calling upon the right onnx dll?

```
Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFCD2E8F2AC: (caller: 00007FFCD2EDEE66) Exception(3) tid(366c) 80070057 The parameter is incorrect.

[2021-04-29 14:36:13.814] [error] [t=13932] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFCD2E8F2AC: (caller: 00007FFCD2EDEE66) Exception(3) tid(366c) 80070057 The parameter is incorrect.

[2021-04-29 14:36:13.815] [error] [t=13932] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```

qm13, 2021-05-03:
@fractalfantasy can you verify you are attempting to execute with DirectML? If so what is the version of Windows you are using.

fractalfantasy, 2021-05-03:
> @fractalfantasy can you verify you are attempting to execute with DirectML? If so what is the version of Windows you are using.

yes trying to initiate the DirectML model
Windows 10 (v.1909)

--

Fyi this is the thread where we are trying to figure it out:
https://github.com/wouterverweirder/kinect-azure/issues/40

qm13, 2021-05-04:
@fractalfantasy looking over this thread is it correct that you can get k4abt_simple_3d_viewer.exe running with DirectML but cannot get your project running?
If that is the case can you please verify you have the correct DLLs in place for your project.

fractalfantasy, 2021-05-04:
Yes k4abt_simple_3d_viewer.exe runs with DirectML fine.

Not sure if you saw the thread I linked, we’ve been trying to update the kinect-azure node but lack of NuGet is making things difficult, still figuring out how to get the .dll referenced properly.

Are you planning on adding NuGet in the future or is it not possible with this architecture?

ZhenyueQin, 2021-05-10:
I believe I am facing the same problem. What is your GPU?

qm13, 2021-05-10:
All, we are working in a new NuGet package which will be released soon. This package will NOT include any of the NVIDIA required DLLs. It will also include only the normal model. NuGet packages are limited to 250MB in size and many of the NVIDIA DLLs have ballooned to far exceed this limit.

closed 2021-08-16 by qm13