microsoft/Azure-Kinect-Sensor-SDK #1563: Unity 3D is freezing while trying to close it

cavisions, 2021-04-12:
The new Body Tracking SDK (1.1.0) seems keeping (for the Unity 3D applications) the sensor connection alive preventing the application from shutting down. Any help would be appreciated. We have been using the older SDK without such complications.

### Setup

- Microsoft Windows [Version 10.0.19042.870]
- VS2019 Community v16.9.3 building 64-bit
- Sensor SDK 1.4.1
- Body Track SDK 1.1.0
- App of your choice; I used the https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking upgraded to Body Tracking SDK 1.1.0 (I can share the upgraded sample if needed)
- Intel i7 with 32GB ram and RTX2060

**To Reproduce in Unity**

1. Start application that uses Kinect Azure
2. Stop the application in Unity
3. Try to close Unity
4. Unity is not responding

**To Reproduce in build**

1. Start application that uses Kinect Azure
2. Try to close the application
3. Application is not responding

cavisions, 2021-04-15:
Have you reproduced this issue? Is there any workaround for this? I'd appreciate some info, because I need to decide whether revert my project to the older SDK or is it worth waiting. Thank you in advance!

anantley, 2021-04-16:
Hi @cavisions,  
I am trying to repro now.  Where can I get access to the 1.1.0 Body Tracking?  Officially I only see 1.0.1
Thanks, Angus

muadiber, 2021-04-26:
I have the same problem

cavisions, 2021-04-26:
@anantley Are there any news here? Thanks in advance!

anantley, 2021-05-18:
@cavisions, there is a fix now for the threading problem, with an updated readme and batch file.
https://github.com/microsoft/Azure-Kinect-Samples/pull/61

seph14, 2021-05-30:
still crash on quit of standalone build.

tested unity 2019.1.2f1 / 2020.3.0 / 2020.3.1, windows10 personal / enterprise.

anantley, 2021-06-01:
@seph14  I just did a fresh git and followed the directions from the readme and could not repro the bug.
Unity 2019.1.2  

I did have some trouble getting the stand alone to run, but here are the dlls in the exe folder that worked for me
[image]

I had to close explorer and reopen it to get the dlls to register.  But it had no problem closing the app while the body tracking was running.

Do you have a crash log?  Or any more config details.  ie visual studio version?

seph14, 2021-06-01:
@anantley that are the dlls I have. Just tested and still crashing. 

Nothing meaningful in the log, only related to this might be the following lines:
```
Setting up 3 worker threads for Enlighten.
  Thread -> id: 3db4 -> priority: 1 
  Thread -> id: 380c -> priority: 1 
  Thread -> id: 26b0 -> priority: 1 
```
These are the last lines in the log.

I changed process mode in SkeletalTrackingProvider to
`ProcessingMode = TrackerProcessingMode.Cuda`
But TrackerProcessingMode.GPU crashes for me too.

Other configs:
VS2017 (15.9.16) (which I think it's irrelevant because Unity build is using their own compiler no?)
script backend:  Mono
API level:            .net standard 2.0
incremental GC: yes

closed 2021-08-16 by qm13