microsoft/Azure-Kinect-Sensor-SDK #1564: Azure Kinect out of stock

catid, 2021-04-13:
Cannot order anymore here: https://www.microsoft.com/en-us/p/azure-kinect-dk/8pp5vxmd9nhq

Is this product end of life, or will there be more inventory?

senshin998, 2021-04-14:
Is there an answer?

Zaptu3117, 2021-04-14:
looking for an answer too

DrMaxP, 2021-04-15:
See https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1531

Zaptu3117, 2021-04-15:
Nobody knows where to find used ones ?

qm13, 2021-04-15:
Duplicate of #1531

closed 2021-04-15 by qm13

JimCheong, 2021-08-31:
I am selling mine in ebay for those that can't wait for $800. Free shipping to Continent of US only. 1st come 1st serve

https://www.ebay.com/itm/255114864826

JimCheong, 2021-08-31:
Item sold - thanks to free market supply meets demand, another developer can continue their work

isabelarboleda28, 2021-09-10:
I am also selling mine on eBay. Here is the link. https://www.ebay.com/itm/185043876603?mkevt=1&mkcid=1&mkrid=711-53200-19255-0&campid=5338273189&customid=&toolid=10001&pub=5575378759&icep_ff3=1&ipn=psmain&icep_vectorid=229466&kwid=902099&mtid=824&kw=lg

JimCheong, 2021-11-19:
> […](#)

Got an email from them today too, but they do not deliver to USA thou.