microsoft/Azure-Kinect-Sensor-SDK #1570: Influence of object colour on depth accuracy

Daniel4144, 2021-04-15:
Hi,
I have a question regarding the accuracy of the Azure Kinect:

I compared the height of lego bricks with different colours and noticed a significant difference there.
The height of the bricks from the desk surface should be approx. 16mm, but the measured depth is wrong with red and black bricks.

For black bricks, the distance is about right, if the brick is in the center of the depth map. With higher distance from the center, the Kinect measures the brick thinner (I only measure the difference between desk depth and brick depth), up to 10mm error before the depth on the black brick becomes invalid (which is expected, because of the low reflectivity of the black).
For red bricks, the error is about 10mm (height ~6mm measured) and remains approximately the same regardless of its position in the depth map.
Other colours I tested, were close to the correct value (white, yellow, blue, grey); grey and blue gave me the best results.
Since the only variable I changed is the colour (same material, same environment), the error somehow has to be correlated to the colour.

Although the error is within the specification of the systematic error, I wonder what the reason could be for that behaviour. According to my (admittedly limited) understanding of time-of-flight sensors I would expect the depth to be more noisy instead of an offset, if a colour reflects less IR light back to the sensor.

I appreciate any ideas or links that could help me to understand the problem.

guanming001, 2021-04-15:
@Daniel4144 you may want to refer to this explanation which mention red lego brick https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1341#issuecomment-687933620

Daniel4144, 2021-04-16:
Thank you @guanming001, I did not expect the same material to have completely different properties for IR light depending on the color.
Since the difference is impossible to correct without prior knowledge of the material, a confidence map as requested in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1408 would be great.

jaygullapalli, 2021-05-06:
Hello, one of our team members looked into a similar issue in the past. This is the summary - The issue you notice is due to [Subsurface scattering](https://en.wikipedia.org/wiki/Subsurface_scattering). Certain colors of Lego block (such as red) are translucent in IR light, which means that there isn’t really a clearly defined surface in the IR for the time-of-flight measurement to range to. The measured path length is substantially further than the physical surface of the object. It’s not inter-object Multi Path Interference that is the primary problem, although the specularity of the blocks can be a problem. Human skin suffers from a similar IR translucency problem, although it is typically less obvious.

closed 2021-07-08 by qm13