microsoft/Azure-Kinect-Sensor-SDK #1571: [request] add python api for k4arecord and k4abt

UnaNancyOwen, 2021-04-16:
**Is your feature request related to a problem? Please describe.**

the python bindings work very fine. 👍 
however, it still doesn't fulfill all features of sdk.
please add k4arecord (record/playback)  and k4abt (bodytracking). thanks,
@JonathanESantos do you have a plan about this in future?

**Describe the solution you'd like**

* add k4arecord (record/playback) to python bindings in azure kinect sensor sdk.
* add k4abt (bodytracking) to python bindigns in azure kinect body tracking sdk.

**Describe alternatives you've considered**

**Additional context**

rida-xavor, 2021-10-08:
@UnaNancyOwen Hi have you tried using K4a_calibration in python or know how can we achieve it? Basically converting Pixel to world coordinates.

UnaNancyOwen, 2021-10-08:
@rida-xavor I has tried converting to point cloud from depth image. My sample program is here.
https://github.com/UnaNancyOwen/AzureKinectSample/blob/master/sample/python/point_cloud.py

In the Python API, all pattern transformations (including per-pixel conversion) are implemented in the transformation class.
These are versions that have not yet been released and may change in the future. If you want more information, Please read source code.
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/python/k4a/src/k4a/_bindings/transformation.py

@JonathanESantos By the way, per-pixel conversion (such as convert_2d_to_3d, convert_3d_to_2d) is included in calibration class in C, C++, and C# APIs. The current implementation of Python API is a little different from other languages. In Python API, these APIs implemented in transformation class. Unless there is a specific problem in Python, I think it should be unified.

UnaNancyOwen, 2021-10-14:
@AlexJian1086 Azure Kinect have depth sensor. You can get depth from depth image. (I don't think we should continue discussion about your question here. @rida-xavor has created a new [issue](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1689) similar to your question, so please discuss it there. this question has nothing to do with this issue.)

zhangzscn, 2021-11-16:
any update for body tracking python sdk？ @qm13