microsoft/Azure-Kinect-Sensor-SDK #1573: Body Track v1.1.0 `frame::get_body()` has wrong exception declaration

diablodale, 2021-04-17:
In body track v1.1.0 the api `k4abt::frame::get_body()` is declared `noexcept`. However, that is invalid given the code. That api *can* throw because it calls `get_body_skeleton()` which can throw. All below found with MSVC code analyzer -- I recommend you include it in your build pipeline too 😉

### Setup

* Body Tracking SDK v1.1.0

### Repro

1. Code review `Azure-Kinect-Body-Tracking-SDK\sdk\include\k4abt.hpp`

### Result

Line 205 is invalid:  `k4abt_body_t get_body(uint32_t index) const noexcept`
because in that function is the call `get_body_skeleton(index, body.skeleton);`
and that function throws.

### Expected

API exception signature to match its actual exception behavior.

### Fix

Somewhat trivial and multiple options. I prefer option 1.

1. Change signature to `k4abt_body_t get_body(uint32_t index) const`
2. Surround call to get_body_skeleton() with try/catch

### Sidebar

MSVC code analyzer also found 6 lines where your variables can be const since they are declared only one and never changed.
```
warning C26496: The variable 'xxxxxx' does not change after construction, mark it as const (con.4).
```

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13

diablodale, 2022-05-25:
@qm13, Fixed in what version? For customers to verify unseeable code changes, we need to know the exact version in which is was potentially fixed so that customers can verify, escalate non-fix, etc.