microsoft/Azure-Kinect-Sensor-SDK #1576: body track v1.1.0 has slow memory leak with Tensor and Tensor-fp16 runtimes

diablodale, 2021-04-20:
I measure a slow memory leak when BodyTrack v1.1.0 is used with the `tensor` and `tensor+fp16` runtimes.
It is approximately 1GB every 45 minutes of continuous 30fps use.

### Setup

* Microsoft Windows [Version 10.0.19042.928]
* Intel(R) Core(TM) i7-10875H CPU @ 2.30GHz, 8c/16t
* NVIDIA GeForce RTX 2070 Super w/ 8GB, and integrated Intel UHD Graphics
* 32GB ram, 1tb disk
* VS2019 Community v16.9.3
* K4A Sensor SDK v1.4.1 and BodyTrack v1.1.0

### Repo

1. Make an app that pulls/gets at 30fps all this data simultaneously: NFOV_UNBINNED depth, 4K color, ir, bodyindex, and skeletons with the regular non-lite onnx model.
2. Enable the app for easy switching between all the runtimes: cuda, directml, tensor, and tensor+fp16
3. Get all of the buffers and body data. Display something if you want.
4. Make a release build
5. Choose CUDA runtime.
6. Run it continuously at 30fps for 90 minutes and watch Task Manager/performance/memory/in-use
7. Repeat previous two steps for all the runtimes.  

### Result

* Tensor and tensor+fp16 leak about 1GB every 45 minutes. The memory in-use gradually increases. In my test of
  tensor+fp16 for 1.5 hours, I also noticed the fps decreased from 30fps->27fps with a noticeable multiple frame latency.
* CUDA shows no significant leak. In my test it was less than 0.1 GB which in this inexact test is insignificant.
* DirectML shows no significant leak.

Runtime | 0 minutes | 15 minutes | 30 minutes | 45 minutes | 90 minutes
-- | -- | -- | -- | -- | --
cuda | 12.5 |   | 12.6 |   |  
directml | 12.8 | 12.8 | 12.8 |   |  
tensor | 12.9 | 13.4 |   | 14.1 |  
tensor+fp16 | 13.2 |   |   |   | 15.9

[image]

### Expected

No leaks.

diablodale, 2021-04-22:
Is it known if the leak is in ONNX code (Microsoft, opensource-ish on github), TensorRT open-source code on github, or Tensor closed-source code owned by NVIDIA?

qm13, 2021-04-23:
Checking with the ONNX team.

stevenlix, 2021-04-23:
Could you try OnnxRuntime master? We've fixed a couple of memory leak issues lately.

diablodale, 2021-04-23:
I (a customer) am not able. The Azure Kinect Body Tracking code is closed-sourced and I have no source code or build scripts in which to compile any part of it. It is 2.5GB spread across 19 files.

Three of the files are named `onnxruntime.dll`,  `onnxruntime_providers_shared.dll`,  `onnxruntime_providers_tensorrt.dll`. They appear to be custom compiles with no version stamps.

qm13, 2021-05-25:
We have verified that the memory leak is in the ONNX Runtime. We have also verified that it is fixed in master. ORT v1.8 is expected to release in June and we plan to update the BT SDK to use ORT v1.8.

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13

diablodale, 2022-05-25:
Fixed in what version? For customers to verify unseeable code changes, we need to know the exact version in which is was potentially fixed so that customers can verify, escalate non-fix, etc.