microsoft/Azure-Kinect-Sensor-SDK #1577: FPS decrease when connect multiple devices

LuyaooChen, 2021-04-20:
When I connect 4 Azure Kinect with one pc, the FPS would decrease to around 10 . I have connect them with different USB port. And it works well when only connect one or two device (30 FPS).  
It seem to because of my pc's performance. Any advise to avoid this problem?

diablodale, 2021-04-22:
First, congrats on having 4 Azure Kinects. I have European customers that would like to buy so many. (they're out of stock)

Next, is your test application single threaded and saturating a single core? Modern CPUs don't have high frequencies. Instead they have many cores. As you process more data, it is important that your processing can run across the many cores in your computer.

For example, the color conversion from HD or higher resolution (MJPEG format) to RBGA is very expensive. If you do this type of processing (plus IR, depth, pointclouds, etc.) it is important that your application code is multithreaded and can scale across many cores so you can process the data simultaneously from multiple Kinect sensors.

LuyaooChen, 2021-04-23:
@diablodale  Thanks for your reply！  
In my application, I create threads for each device. My code looks like this:
```
void run()
{
         // thread for each devcie
        dev.get_capture(&capture);
        // without doing data processing now.
}
int main()
{
    //.... 
    //main thread
    while(1)
    {
            QTime time;          // btw, I use QT
            time.start(); 

            for(int i=0;i<4;i++) devs[i].start(); // start running the subthread(non-blocking) and it will call run().
            while(1) {/*wait for all threads finished*/}

            cout<<time.elapsed()<<"ms"<<endl;    // time for get_capture()
    }
}
```
I confirmed this question again, and I found that the FPS would decrease to 15 with 2 devices, and about 10-12 with 3 devices, and 7-9 with 4 devices. 
They are synchronized using 3.5mm synchronization port (it's similar if running with standalone mode).
And I have tested on MJPEG and BGRA format, the FPS are no difference. 
Other configs for Azure Kinect is:
```
config.camera_fps = K4A_FRAMES_PER_SECOND_30;
config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
config.color_resolution = K4A_COLOR_RESOLUTION_720P;
config.synchronized_images_only = true;
```
My computer's environment: Intel E5-2678, 16GB memory, RTX 2070S GPU, Ubuntu 18.04 OS.

diablodale, 2021-04-23:
Hmm. Something seems suspicious. A Kinect can fastest produce 30fps...
30 / 2 = 15
30 / 3 = 10
30 / 4 = 7.5
These are all the same fps that you wrote. My instinct tells me there is some code somewhere that operates at a maximum of 30fps. As you add more Kinect sensors, it divides its work evenly across the Kinects. This is usually code that is associated with a timer, something periodic, thread pool managers, etc. Or it might be related to synchronization.

Synchronizing all the Kinects is not trivial. You will need to start the sensors in a specific order, and control the laser timing offset of each Kinect. If you don't manage these things, then you can likely have many problems including decreased fps due to interference/conflicts. The Kinect docs have 2 or 3 pages that discuss these topics. Plus a few good github issues to read like...
https://docs.microsoft.com/en-US/azure/Kinect-dk/multi-camera-sync
https://docs.microsoft.com/en-us/azure/kinect-dk/capture-device-synchronization
https://docs.microsoft.com/en-us/azure/kinect-dk/record-external-synchronized-units
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1240

While you are testing, I recommend you change to K4A_IMAGE_FORMAT_COLOR_NV12 and never get any images; don't call get_xxxxx_image(). Focus on only calling get_capture() so that you can start to isolate the cause.

LuyaooChen, 2021-04-26:
@diablodale I tried to test sensors using k4aviewer, and this problem still exist. BUT it can work on another computer!  So I think this problem is caused by hardware.
Thanks for your help again.

closed 2021-04-26 by LuyaooChen