microsoft/Azure-Kinect-Sensor-SDK #1578: Color Camera: received malformed image data

l1346792580123, 2021-04-20:
**Describe the bug**

When I open the k4aviewer to view the data captured by azure kinect, the color camera has the following error: Color Camera: received malformed image data. Then I can not view the color image any more. How can I solve this problem?
**To Reproduce**
open k4aviewer.exe
open device
set depth mode to WFOV_Unbinned
set color format mjpg
set color resolution 1080p
set framerate 15FPS
start

**Expected behavior**
display color image

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 19041.867
 - SDK Version: 1.4.1
 - Firmware version: Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

**Additional context**

qm13, 2021-04-23:
Can you verify whether the color camera works as webcam?

l1346792580123, 2021-04-28:
Sorry for the late reply.  I have tested other format like BGRA, NV12 and YUV12, and these formats can work properly. And I have found that there are some warnings will appear when getting the above error. Does it matter?
[image]

qm13, 2021-04-28:
@l1346792580123 these warning indicate that the GPU did not have enough cycles to complete processing the raw data from the camera. If this persists your GPU is not powerful enough for the combined workloads you are putting on it.

crystal-butler, 2021-05-10:
I am getting the "malformed image data" error originally described by the reporter of this bug when viewing or recording the color camera stream, but under the following environment:
Platform & OS: ARM, NVIDIA Jetson Xavier NX, Linux Ubuntu 18.04 (4.9.201-tegra)
SDK Version: 1.4.1
Firmware Versions:
- RGB camera firmware: 1.6.110
- Depth camera firmware: 1.6.79

This issue occurs with all combinations of fps and resolution under MJPG format when using k4aviewer live, although at 5 fps and 720p the error takes longer to show up (around 30 seconds vs. 1-15 seconds for higher res & frame rates). If I change the format to BGRA, the color camera stream continues to display, but with periodic warnings:
`DecodeMJPEGtoBGRA32(). MJPEG decode failed, dropping image: -1`

Recordings made using k4arecorder also fail to display the color camera stream when opened in k4aviewer, although no errors are reported during the recording process. The error that occurs during playback in k4aviewer is "Failed to get next capture!" It isn't possible to set k4arecorder to use the BGRA format that was working with warnings in k4aviewer for comparison.

The Kinect I'm working with does not have this problem when connected to an Ubuntu 18.04 laptop running the Sensor SDK, so presumably it is something to do with the SDK for ARM or the Jetson's capabilities, combined with error handling in the SDK. If the color camera stream can continue with warnings under the BGRA format, can't it do the same for MJPEG, which doesn't require a change of encoding?

qm13, 2021-05-11:
@crystal-butler there are two problems here 1) you are getting bad frames in the recording; and 2) k4aviewer stops when it encounters a bad frame. We agree that k4viewer should skip bad frames rather than stopping play back (happy yo take a PR to fix the k4viewer). The getting bad frames in the recording it likely due to either the USB or drive performance.

jcova1996, 2021-07-12:
I also seem to have issues using the mjpg color format. I'm only able to catch 10-15 frames from a device in mjpg before the captures no longer contain color information. If I run the same code with bgra color format, I don't get these issues.