microsoft/Azure-Kinect-Sensor-SDK #1579: k4abt_tracker_create fails because of no CUDA, even when running in CPU mode.

jbrownkramer, 2021-04-20:
**Describe the bug**

k4abt_tracker_create(&calib, tracker_config, &tracker); returns K4A_RESULT_FAILED. and the logs indicate the problem is a missing CUDA dll, but I am targeting the CPU or integrated GPU, so I shouldn't need CUDA

Note, I don't think this is relevant, but I am trying to use k4abt_tracker_t on recorded data.  calib is created from k4a_calibration_get_from_raw with the calibration data being read from disk.  tracker_config as below:

**To Reproduce**

```
    k4abt_tracker_configuration_t tracker_config = K4ABT_TRACKER_CONFIG_DEFAULT;
    tracker_config.processing_mode = K4ABT_TRACKER_PROCESSING_MODE_CPU;
    tracker_config.model_path = "pathtomodel\\dnn_model_2_0_op11.onnx";
    k4abt_tracker_t tracker = NULL;
    k4a_result_t result =  k4abt_tracker_create(&calib, tracker_config, &tracker);
```

**Expected behavior**

I would not expect an error, and I would expect tracker to not be null.  I would expect the CUDA requirement only when using CUDA.

**Logs**

```
New logging session started

[2021-04-20 16:00:36.833] [info] [t=13320] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (493): VerifyDependencies(). Verifying Dependencies!
[2021-04-20 16:00:36.835] [error] [t=13320] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (500): VerifyDependencies(). Cannot locate cudart64_110.dll; Please install CUDA 11.1 at https://developer.nvidia.com/cuda-11.1.0-download-archive
[2021-04-20 16:00:36.835] [error] [t=13320] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (154): Create(). At least one dependent library is missing!
[2021-04-20 16:00:36.835] [error] [t=13320] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Pro 10.0.19042 Build 19042
 - SDK Version:  1.1.0
 - Firmware version:  Running from recording, no firmware version recorded.

jbrownkramer, 2021-04-21:
One update: I installed CUDA (though I shouldn't have to), and now I have this error:

```
New logging session started

[2021-04-21 14:59:48.891] [info] [t=25672] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (493): VerifyDependencies(). Verifying Dependencies!
[2021-04-21 14:59:48.892] [info] [t=25672] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (171): Create(). Initializing CUDA and ONNX Runtime... 
[2021-04-21 15:00:24.307] [error] [t=25672] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Find onnxruntime.dll at C:\Users\jbrownkramer\projects\ComputerVision\DontPanic.CV.Tracking\DontPanic.CV.Clients.DebugClient\bin\x64\Debug\net461\onnxruntime.dll but it doesn't load correctly!
[2021-04-21 15:00:24.307] [error] [t=25672] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```

jbrownkramer, 2021-04-21:
Following up on my most recent post: I ran onnxruntime.dll through dependency walker, and it itself has a dependency on cudnn64_8.dll.  Throwing that in with the build folder, and it now works.  It would be best for us (to keep deployment size down if nothing else) if this dependency didn't exist when we're not targeting an NVIDIA GPU.

diablodale, 2021-04-22:
seems to be related or a dup of https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1154

qm13, 2021-04-23:
Dup of #1154 Fixed in the next release of ONNX runtime.

closed 2021-04-23 by qm13