microsoft/Azure-Kinect-Sensor-SDK #1580: Body Tracking Comments Indicate Only Depth Required, but IR Seemingly Required in Practice

jbrownkramer, 2021-04-22:
**Describe the bug**
We store only depth data, and we are trying to use body tracking on our recorded data.  I have populated a capture object with the depth image, but I have run into some problems.

The documentation for k4abt_tracker_enqueue_capture in k4abt.h says:

```
* \param sensor_capture_handle
* Handle to a sensor capture returned by k4a_device_get_capture() from k4a SDK. It should contain the depth data for
* this function to work. Otherwise the function will return failure.
```

However, when I create a capture object without populating the IR image, and try to call k4abt_tracker_enqueue_capture, I get the following error in the logs:

`[2021-04-22 11:29:45.690] [error] [t=27804] [K4ABT] ..\src\TrackerHost\DepthFrameBlobK4A.cpp (19): Initialize(). Get IR buffer from the capture handle failed!`

Furthermore, if I reuse the depth image to populate the IR image as well, I don't get the error, but k4abt_frame_get_num_bodies(body_frame) always returns 0, even if there are people walking around unobstructed in the frame.

**To Reproduce**
Here's some code we're trying

```
    k4a_capture_t capture;
    k4a_capture_create(&capture);
   
    pin_ptr<unsigned short> op = &frame->DepthFrame->RawData[0]; //This is our format.  Note, it's stored in memory in the same format as a the images coming out of a live camera.
    unsigned short* rawData = op;
    uint8_t* buffer = (uint8_t*)rawData;

    int w = frame->DepthFrame->Width;
    int h = frame->DepthFrame->Height;

    uint8_t* irBuffer = new uint8_t[2 * w * h];
    for (int i = 0; i < 2 * w * h; i++)
    {
        irBuffer[i] = 0;
    }

    k4a_image_t depth_image;
    k4a_image_create_from_buffer(K4A_IMAGE_FORMAT_DEPTH16, w, h, 2 * w, buffer, 2 * w * h, NULL, NULL, &depth_image);
    k4a_capture_set_depth_image(capture, depth_image);

    k4a_image_t ir_image;
    k4a_image_create_from_buffer(K4A_IMAGE_FORMAT_IR16, w, h, 2 * w, irBuffer, 2 * w * h, NULL, NULL, &ir_image);
    k4a_capture_set_ir_image(capture, ir_image);

    k4a_wait_result_t queue_capture_result = k4abt_tracker_enqueue_capture(this->tracker, capture, K4A_WAIT_INFINITE); 

    k4abt_frame_t body_frame = NULL;
    k4a_wait_result_t pop_frame_result = k4abt_tracker_pop_result(this->tracker, &body_frame, K4A_WAIT_INFINITE);

    k4a_capture_release(capture);

    if (pop_frame_result == K4A_WAIT_RESULT_SUCCEEDED)
    {
        size_t num_bodies = k4abt_frame_get_num_bodies(body_frame);  //Always 0

        k4abt_frame_release(body_frame); // Remember to release the body frame once you finish using it
    }
```

**Expected behavior**
I expect body tracking to work on our recorded data, even if the IR frame is not populated.

**Logs**
After working through various issues, there are no logged errors.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Pro 10.0.19042 Build 19042
 - SDK Version:  1.1.0
 - Firmware version:  Running from recording, no firmware version recorded.

conatsera, 2021-04-23:
https://www.microsoft.com/en-us/research/uploads/prod/2020/01/AKBTSDK.pdf 
Page 11 describes the body tracking system architecture. The IR image is processed to create the body segmentation map and create a 2D joint map.

jbrownkramer, 2021-04-23:
Ah, thank you, that answers my question!

qm13, 2021-04-26:
@conatsera is correct. Body tracking requires both IR and depth images as part of the capture.

closed 2021-04-26 by qm13

alpha3d-bin, 2024-04-07:
[2024-04-07 14:34:13.352] [error] [t=14756] [K4ABT] ..\src\TrackerHost\DepthFrameBlobK4A.cpp (15): Initialize(). Depth buffer size doesn't match expected image size!
[2024-04-07 14:34:13.352] [error] [t=14756] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (274): EnqueueCapture(). Initialize DepthFrameBlob failed!