microsoft/Azure-Kinect-Sensor-SDK #1588: Frames with varying color/depth shifts (some frames are fine, others are noticeably off)

meiler-inunum, 2021-05-03:
**Description**
Some frames show a shift between the color and depth information. Our suspicion is that
this is not a calibration issue, since it is not consistent. Some frames are fine, others aren't.
The following images show two frames recorded within half a second of each other and almost no movement in between.
We also had a look at the timestamps sent with the depth and image data which were within some microseconds.
The option synchronized_images_only is enabled on the api.

An image of the scene and the colorised pointclouds of two frames:
[https://imgur.com/a/wh5PuJI](https://imgur.com/a/wh5PuJI)

We also took all the calibration data and projected the data ourselfes with the same results.
(Intrinsics & Extrinsics applied, as well as undistortion applied)

**Hardware:**
 - OS with Version: Windows 10
 - SDK Version:  1.4.0
 - Firmware version: 
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

I noticed that there is a firmware update. I will try to get access to the warehouse one more time and do the same tests again. I will let you know in case anything changes.

I also read through the other issues regarding similar issues. Although in those cases they documented a general shift which could be fixed by calibration. As mentioned in the beginning I think this is different as we cannot get consistent results.

meiler-inunum, 2021-05-04:
No change in the behaviour after the firmware update to 1.4.1.

I uploaded a recording which again shows the effect quite well:
[https://www.swisstransfer.com/d/02ad2696-6373-4ff9-98d8-abc730e2adcb](https://www.swisstransfer.com/d/02ad2696-6373-4ff9-98d8-abc730e2adcb)

If you look at the frame with timestamp 20011866 you will see a significant shift in the color layer on the edge of the meeting room ceiling. Whereas in a frame like 17478533 it is perfectly fine. I also uploaded the two exported frames as LAS files so that you can easily have a look at the issue in something like cloudcompare:

[https://www.swisstransfer.com/d/1c10f6d8-9b9f-4cd0-b58d-7d197bd98d78](https://www.swisstransfer.com/d/1c10f6d8-9b9f-4cd0-b58d-7d197bd98d78)

Thanks in advance for having a look at it!

qm13, 2021-05-04:
@meiler-inunum  sorry, what tool should I use to view .las files?
I cannot see the issue in the two frames you reference. I assume by color layer you are referring to the colorization of the depth image and not the RGB camera image?
Couple of other points: the depth camera uses modulated IR to measure distance which can be affected by direct sunlight through the window, and you can also get a multi-path affect on edges between two intersecting surfaces.
Have you tested with a second Azure Kinect?

meiler-inunum, 2021-05-05:
Thanks for your response! You can open the pointcloud files in CloudCompare: [https://www.danielgm.net/cc/](https://www.danielgm.net/cc/) (a well known open source pointcloud/mesh application).

I took two screenshots and marked the areas of interest:
[https://imgur.com/a/W4v1T9d](https://imgur.com/a/W4v1T9d)

As you see in the first image the color is shifted. Some pixels representing the wall are actually used to inpaint the ceiling.

Regarding your questions: Correct, I'm talking about the inpainted depth image. The effects we are seeing are incorrectly inpainted depth data - because the depth layer and the color layer itself are (except for the colors position) correct on their own. That's why I said it looks like those two "images" are not taken at exactly the same time.
We have two Azure Kinect devices, both showing the same issue.

Best,
Michael

qm13, 2021-05-05:
@meiler-inunum It sounds like you mean the RGB overlaid on depth point cloud (3D button in k4aviewer).
This looks suspiciously like multi-path were the IR light is returning to the sensor via more than one path. This manifests as bowing of edges. Moving the sensor will change the path(s) resulting in changing depth values. The RGB image is mapped onto the depth image so if the depth is wrong the resulting RGB+depth point cloud will also be wrong.

meiler-inunum, 2021-05-06:
I think I should elaborate a bit more about our setup. Maybe this will help to clarify any misunderstandings.

We use the Azure Kinect as a sensor and process all the data in our own framework. This contains a registration algorithm building up a pointcloud. Meaning we end up with a pointcloud of the whole scene and a trajectory. This trajectory will then be used to project the color information onto the pointcloud. If the color image and the depth image have been recorded at exactly the same time ( a few microseconds difference shouldn't matter) the colorised pointcloud should look perfectly fine. Unfortunately we noticed some pictures which were still off after all our optimizations which we did internally already. Due to this we tried to further investigate and ended up at the sensor level. In order to figure out which part has caused the shift we checked the data we get from the k4a sdk and colorised the frames as we get them in. After all the color values there should perfectly overlap with the depth data (of course after applying undistortion and extrinsics/intrinsics). Unfortunately for some images (like the ones I linked to above) this doesn't seem to work as expected, the color information is shifted compared to the depth layer. The pointcloud and the color layer simply do not match. That this is on an edge is just because it was easier to show the symptom, but the shift is through the whole frame.

Here I have another example of a scan having tthis issue. The pointcloud has been converted to a mesh and we used multiple images to texturize it (for visualization purposes, the base issue is the same):
[https://imgur.com/a/9tDr6lp](https://imgur.com/a/9tDr6lp)
You can see a shift between almost every image. Obviously we are aware that the calibration and trajectory aren't perfect either, but in that case the errors should at least be predictable and not visible in the captures we get from the sdk on their own.

Does this help to describe our issue in a better way?

meiler-inunum, 2021-05-11:
I'm not explicitly using it, but neither did I set anything to disable whatever might be running on the device. I didn't find anything on the k4a api to disable the audio parts. Do we have to use another api to disable it entirely?

qm13, 2021-05-11:
Sorry for the confusion. AE = Auto Exposure. Are you using auto exposure or manual exposure?

meiler-inunum, 2021-05-12:
Not sure about the scan I uploaded but in general I disable it to get more consistent images.

closed 2021-10-06 by qm13