microsoft/Azure-Kinect-Sensor-SDK #1589: missing firmware

christian-rauch, 2021-05-03:
**Describe the bug**
Firmware files are missing.

There are no firmware files in the source or build directory that have a filename pattern like `AzureKinectDK*.bin`. The folder `firmware` only contains a source to build the library `libk4a_firmware.a`.

**To Reproduce**
1. download and build SDK from https://github.com/microsoft/Azure-Kinect-Sensor-SDK
2. search for firmware files: `find -name *.bin`

**Expected behaviour**
At least one firmware file should be discovered, e.g. `AzureKinectDK_Fw_1.5.926614.bin` as described in https://docs.microsoft.com/en-us/azure/kinect-dk/update-device-firmware

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 20.04
 - SDK Version:  git master
 - Firmware version: ???

qm13, 2021-05-03:
The Azure Kinect Sensor SDK does not currently support Ubuntu 20.04. Support is included in the next update.

christian-rauch, 2021-05-03:
The firmware will still be missing if I clone and build this repo on Ubuntu 18.04, I assume.

Or can you point me to where the binary firmware is within this repo, or where it is built?

HlibKazakov2000, 2021-05-04:
I have checked the content of both `libk4a1.4` and `libk4a1.4-dev` packages using `apt-file list` and it seems like the firmware was not included in any of them.
We will keep you posted.

christian-rauch, 2021-07-15:
Note to others landing on this page to get the binary firmware: The firmware itself is directly available from https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md. No need to set up a Windows VM to install the SDK and extract the firmware image.

closed 2021-12-09 by qm13