microsoft/Azure-Kinect-Sensor-SDK #1590: failed to detect a second device after an attempt to open both

christian-rauch, 2021-05-03:
**Describe the bug**
After attempting to open 2 devices in parallel via a single USB controller, the SDK is unable to detect the second device. While the original issue is mentioned in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1049, this is a request to fix the SDK error, that prevents using one of the two devices, after opening both devices via a single USB controller in parallel failed.

**To Reproduce**
1. connect 2 Kinect devices
2. open the first device via `k4aviewer`
3. open the second device via `k4aviewer`, this will fail
4. close all `k4aviewer` instances
5. list devices via `AzureKinectFirmwareTool -q`

After this, `AzureKinectFirmwareTool -q` will only show one of the two devices and a lot of error messages.

**Expected behaviour**
`AzureKinectFirmwareTool` should list all two devices:
```
$ AzureKinectFirmwareTool -q
 == Azure Kinect DK Firmware Tool == 
Device Serial Number: 000085101312
Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

Device Serial Number: 000052701312
Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Logs**
After this, I get
```
$ AzureKinectFirmwareTool -q
 == Azure Kinect DK Firmware Tool == 
[2021-05-03 16:35:04.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2021-05-03 16:35:04.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:04.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:04.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length

Device Serial Number: 000085101312
Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

[2021-05-03 16:35:05.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2021-05-03 16:35:05.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:05.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:05.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length

[2021-05-03 16:35:06.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2021-05-03 16:35:06.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:06.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:06.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length

[2021-05-03 16:35:07.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2021-05-03 16:35:07.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:07.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:07.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length

[... 76 lines cut ...]
[2021-05-03 16:35:23.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:23.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:23.459] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length

[2021-05-03 16:35:24.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2021-05-03 16:35:24.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:24.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:24.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length

[2021-05-03 16:35:25.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2021-05-03 16:35:25.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2021-05-03 16:35:25.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2021-05-03 16:35:25.458] [error] [t=1537367] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/firmware/firmware.c (486): firmware_get_serial_number(). Failed to get serial number length
```

**Desktop (please complete the following information):**
 - OS with Version: Ubutnu 20.04
 - SDK Version:  1.4.1
 - Firmware version: 
 ```
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
```

qm13, 2021-05-03:
The Azure Kinect Sensor SDK does not currently support Ubuntu 20.04. Support is included in the next update.

christian-rauch, 2021-05-04:
I am perfectly able to use both Kinects at 720p via the UVC (USB video device class) interface in Ubuntu. E.g. I am able to show both cameras via two instances of `guvcview`.

I believe the limitation is just an issue with this SDK, not with the hardware.

qm13, 2021-05-04:
Could this be a bandwidth issue? Seems from your notes above that the first device works fine in k4aviewer but all fails when you invoke the second instance of k4aviewer.

closed 2021-08-16 by qm13